osresearch/airbreak #4: Communication channel

dbuentello, 2020-04-14:
Is their a Slack/IRC channel for collaboration?

dbuentello, 2020-04-14:
From: https://news.ycombinator.com/item?id=22867960

@wormyrocks :
> If you are a SleepyHead/OSCAR developer, or have access to an AirCurve S10 and an ST programmer, I would love to talk.

FWIW, I have an Airsense 10, Aircurve, and ST programmer (somewhere in my junk drawer). I can assist with your ask.

dbuentello, 2020-04-14:
I'm also working on getting code execution via the SD card. It's still very early and premature. Certain machines (e.g. Resmed S9) will reject SD cards not in the correct format. 

From: http://www.apneaboard.com/wiki/index.php/Running_OSCAR_for_the_first_time
> WARNING: It's wise to make sure the card is locked if you are using a Mac or a Windows 8.1 or Windows 10 machine since these operating systems will write hidden files to unlocked cards. And the ResMed S9 machines will reject SD cards with any extra files written on them. See Secure Digital (SD) memory card for instructions on locking a card.

wormyrocks, 2020-04-15:
Haven't worked out too many collaboration details yet. Shoot me a DM on Twitter? @wrmrx

osresearch, 2020-04-16:
We've setup airbreak-dev.slack.com for discussions. Send an email to info@airbreak.dev for an invite.

closed 2020-04-16 by osresearch