osresearch/airbreak #7: AirStart 10 support

osresearch, 2020-04-16:
Several people have asked about the very low end AirStart 10 support.  I don't have access to one to verify compatibility, but opening this issue for tracking it.

andrewleech, 2020-04-16:
The airstart does not appear to have the control dial on the front which is going to be a limiting factor for usage
https://www.resmed.com/ap/en/consumer/products/devices/airstart-10-cpap.html
Perhaps it's the same circuit board just needing a suitable encoder soldered on, but that's unlikely to be the only difference.

I'm quite interested in support for the airsense elite, it visually looks the same at least.

osresearch, 2020-04-16:
[image]
The AirSense 10 board has two unpopulated switches (SW204 and SW205) that appear to line up with where the AirStart 10 buttons are. My guess (and this is totally a guess!) is that they are the same mainboard with different interface elements.

dakotaKat, 2020-04-16:
Working on getting a prescription updated so I can get an AirStart shipped out to me so I can tear it down and dump the flash.

colinoflynn, 2020-04-16:
@andrewleech do you mean AirSense 10 Elite? There's nothing to suggest there is any difference in HW between the "AirXXXX 10 YYYYY" units at all. I.E., the AirCurve 10 VAuto uses exact same PCB PN as the AirSense 10 CPAP if devices are of same vintage - even devices across large time-spans look to be binary compatible although PCB rev changes.

andrewleech, 2020-04-16:
@colinoflynn yes the airsense elite units are a bit cheaper than the autoset so it'd be a decent candidate. If anyone else tries it, it'd be good to confirm its not missing some of the air flow sensors etc.

colinoflynn, 2020-04-16:
Haven't looked at that specific unit, but I'd note that it's also on the list of labels (see https://fccid.io/2ACHL-AIR10CD/Label/label-info-2328354) under the FCC approval.

alanarthur, 2020-04-21:
I'd be very interested in testing to see if I can 'upgrade' my Airsense 10 Elite to become an Airsense 10 Autoset by replacing the firmware. I.e change it from being a fixed pressure machine to an automatic pressure machine

nimamoslemy, 2020-05-03:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)

ArthurMafra, 2022-03-13:
> Working on getting a prescription updated so I can get an AirStart shipped out to me so I can tear it down and dump the flash.

Hi @dakotaKat , it has been long time that the project has been abandoned. Do you happen to have any Aircurve OEM firmware that I can used to fix Airsense that have been donated to me? If so I really appreciate. Best regards, arthur iphone @ gmail .com

macka-the, 2024-06-23:
Can confirm the Airsense 10 Elite is fully functional as Autoset, ASV and vAuto am currently looking for mainboard from an AirCurve 10 ST-A