osresearch/airbreak #17: AirSense 10 "Autoset For Her" and other device versions support?

dji-transpire, 2020-04-19:
Hi, the firmware differences are obviously holding this project back as there are so many slightly different versions of the device. Are there plans to check whether one device can be operated with a firmware labeled for another device version?

colinoflynn, 2020-04-19:
Devices _appear_ to operate fine with FW from one device flashed onto another (i.e., AirCurve 10 VAuto --> AirSense 10 AutoSet For Her). I had started to collect some results of quickly tested units at https://cpap2bipap.com (but it's just a dumber summary of what Trammell et al. had already discovered, the HW is highly capable and standardized, and Trammell et al. did more verification of pressure outputs over time).

bdarcus, 2020-04-25:
Could you update the docs to reflect the fact that this works for more than just the autoset? Perhaps a FAQ entry with whatever current understanding and caveats are?

Drmaestro, 2020-05-20:
Is the website https://cpap2bipap.com/ still alive?

ArthurMafra, 2022-03-13:
> Devices _appear_ to operate fine with FW from one device flashed onto another (i.e., AirCurve 10 VAuto --> AirSense 10 AutoSet For Her). There are some results of quickly tested units at https://cpap2bipap.com (from another colleague who had suggested the reflash approach, at the time not knowing Trammell et al. had already discovered also that the HW is highly capable, and AirBreak has more verification on unit outputs).

hi @colinoflynn , is it possible for you to share Aircurve firmware. I had mine 306 patched but only mode that works (not sure if correctly) is ASVauto. If you fell that you can help me please use my email arthuriphone @ gmail . com. Airsense and aircurve 10 are to be replaced by 11 models but I cannot afford buy VAuto or ST-A and I have donated Airsense 10. 
Appreciated.