osresearch/airbreak #31: invalid command name "mrh"

nbritton, 2020-04-30:
`
root@ubuntu:~/airbreak# head -n 4 tcl/airsense.cfg
source [find interface/raspberrypi2-native.cfg]
transport select swd
set WORKAREASIZE 0x2000
source [find target/stm32f4x.cfg]

root@ubuntu:~/airbreak# openocd -f ./tcl/airsense.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 2002 kHz
Info : SWD DPIDR 0x2ba01477
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
./tcl/airsense.cfg:78: Error: invalid command name "mrh"
in procedure 'script'
at file "embedded:startup.tcl", line 60
at file "./tcl/airsense.cfg", line 78

root@ubuntu:~/airbreak# cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

root@ubuntu:~/airbreak# uname -a
Linux ubuntu 5.4.0-1008-raspi #8-Ubuntu SMP Wed Apr 8 11:17:03 UTC 2020 armv7l armv7l armv7l GNU/Linux
`

nbritton, 2020-04-30:
It seems an extra library is required, located here: https://github.com/ntfreak/openocd/blob/master/tcl/mem_helper.tcl

wget https://raw.githubusercontent.com/ntfreak/openocd/master/tcl/mem_helper.tcl;

bdarcus, 2020-05-02:
I ran into that issue, but the library is included in more recent versions of openocd. I think I was on 0.10.3. I guess the PR linked here just means this will work with 0.10.0, which is more widely available in package managers?

scaffoo, 2023-09-28:
Sorry if this is not the correct place for this but I'm new to all this - I'm trying to follow the instructions here https://airbreak.dev/firmware/  and I'm using wsl on windows as I don't have a linux machine.  I was able to get the hardware connected, install openocd (version 0.10.0, I don't know how to get another version).  

When I call this command: sudo openocd -f interface/stlink-v2.cfg -f 'tcl/airsense.cfg

I originally got the error: tcl/airsense.cfg:75: Error: invalid command name "mrh"

So I used wget to download mem_helper.tcl.  But now I get:

mem_helper.tcl:7: Error: invalid command name "read_memory"

Is there another tcl file to download or am I doing something more fundamentally wrong like the wrong version of openocd

scaffoo, 2023-09-29:
FYI: I worked around this by using Debian instead of Ubuntu.  When I installed openocd I got version 12 instead of 10 and then it worked.