osresearch/airbreak #33: computed checksum did NOT match

nbritton, 2020-04-30:
root@ubuntu:~/airbreak# ./patch-airsense stm32.bin stm32-unlocked.bin
stm32.bin: FAILED
sha256sum: WARNING: 1 computed checksum did NOT match
stm32.bin: wrong hash

SW firmware version on my device is SX567-0306, it's been unplugged for about six months so I guess it did not receive the over the air updates to version SX567-0401. I don't know how to trigger the OTA firmware update process, I don't have a DME provider so I'm not even sure this unit can receive the update. The machine has one bar on the modem signal status. Is there a way to download the stock firmware update version SX567-0401 on the Internet?

nbritton, 2020-04-30:
I changed the sha256 checksum in the bash script to match my v306 firmware and flashed the device anyways. The flash appears to have worked because my machine now says "HACKED!". However, none of the extra options show up in the mode menu, it just shows CPAP and Autoset.

nbritton, 2020-04-30:
I flashed the machine back to the stock firmware, that appears to have worked as expected... the machine no longer says "HACKED!". I reset the Redmed machine with a power cycle. However, I now get this message on the Raspberry Pi... 

root@ubuntu:~/airbreak# openocd -f ./tcl/airsense.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
Info : replaced existing 'mrw' usage
Info : replaced existing 'mrw' help
Info : replaced existing 'mrb' usage
Info : replaced existing 'mrb' help
Info : replaced existing 'mmw' usage
Info : replaced existing 'mmw' help
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 2002 kHz
in procedure 'init' called at file "./tcl/airsense.cfg", line 77
in procedure 'ocd_bouncer'

nbritton, 2020-04-30:
root@ubuntu:~/airbreak# openocd -f ./tcl/airsense.cfg -d 3
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 5 command.c:544 command_print(): debug_level: 3
Debug: 14 6 options.c:181 add_default_dirs(): bindir=/usr/bin
Debug: 15 6 options.c:182 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 16 6 options.c:183 add_default_dirs(): exepath=/usr/bin
Debug: 17 6 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 6 configuration.c:42 add_script_search_dir(): adding /root/.openocd
Debug: 19 6 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 20 6 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 21 6 configuration.c:82 find_file(): found ./tcl/airsense.cfg
Debug: 22 6 configuration.c:82 find_file(): found /usr/bin/../share/openocd/scripts/interface/raspberrypi2-native.cfg
Debug: 23 6 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface bcm2835gpio
Debug: 24 7 command.c:143 script_debug(): command - interface ocd_interface bcm2835gpio
Debug: 26 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_jtag_nums'...
Debug: 27 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_tck_num' does not have the '.usage' field filled out
Debug: 28 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tck_num'...
Debug: 29 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_tms_num' does not have the '.usage' field filled out
Debug: 30 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tms_num'...
Debug: 31 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_tdo_num' does not have the '.usage' field filled out
Debug: 32 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tdo_num'...
Debug: 33 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_tdi_num' does not have the '.usage' field filled out
Debug: 34 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tdi_num'...
Debug: 35 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swd_nums'...
Debug: 36 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_swclk_num' does not have the '.usage' field filled out
Debug: 37 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swclk_num'...
Debug: 38 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_swdio_num' does not have the '.usage' field filled out
Debug: 39 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swdio_num'...
Debug: 40 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_srst_num' does not have the '.usage' field filled out
Debug: 41 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_srst_num'...
Debug: 42 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_trst_num' does not have the '.usage' field filled out
Debug: 43 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_trst_num'...
Debug: 44 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_speed_coeffs' does not have the '.usage' field filled out
Debug: 45 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_speed_coeffs'...
Debug: 46 7 command.c:319 command_new(): BUG: command 'bcm2835gpio_peripheral_base' does not have the '.usage' field filled out
Debug: 47 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_peripheral_base'...
Debug: 48 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_peripheral_base 0x3F000000
Debug: 49 8 command.c:143 script_debug(): command - bcm2835gpio_peripheral_base ocd_bcm2835gpio_peripheral_base 0x3F000000
Debug: 51 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_speed_coeffs 146203 36
Debug: 52 8 command.c:143 script_debug(): command - bcm2835gpio_speed_coeffs ocd_bcm2835gpio_speed_coeffs 146203 36
Debug: 54 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_swd_nums 25 24
Debug: 55 8 command.c:143 script_debug(): command - bcm2835gpio_swd_nums ocd_bcm2835gpio_swd_nums 25 24
User : 57 8 command.c:544 command_print(): BCM2835 GPIO nums: swclk = 25, swdio = 24
Debug: 58 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_srst_num 18
Debug: 59 8 command.c:143 script_debug(): command - bcm2835gpio_srst_num ocd_bcm2835gpio_srst_num 18
User : 61 8 command.c:544 command_print(): BCM2835 GPIO config: srst = 18
Debug: 62 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_reset_config srst_only srst_push_pull
Debug: 63 8 command.c:143 script_debug(): command - reset_config ocd_reset_config srst_only srst_push_pull
User : 65 8 command.c:544 command_print(): srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Debug: 66 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select swd
Debug: 67 8 command.c:143 script_debug(): command - ocd_transport ocd_transport select swd
Debug: 68 8 command.c:319 command_new(): BUG: command 'swd' does not have the '.usage' field filled out
Debug: 69 8 command.c:364 register_command_handler(): registering 'ocd_swd'...
Debug: 70 8 bitbang.c:353 bitbang_swd_init(): bitbang_swd_init
[... cut ...]
Debug: 307 22 bitbang.c:531 bitbang_swd_run_queue(): bitbang_swd_run_queue
Debug: 308 22 bitbang.c:360 bitbang_exchange(): bitbang_exchange
Debug: 309 22 bitbang.c:538 bitbang_swd_run_queue(): SWD queue return value: 07
Debug: 310 22 command.c:626 run_command(): Command failed with error code 7
User : 311 22 command.c:687 command_run_line(): in procedure 'init' called at file "./tcl/airsense.cfg", line 78
in procedure 'ocd_bouncer'
Debug: 312 22 command.c:626 run_command(): Command failed with error code -4
User : 313 23 command.c:687 command_run_line():

nbritton, 2020-04-30:
I very briefly had it working after I shorted RST to v3.3 (R-Pi pin 1)... the following was output to the terminal... I'm wondering if maybe I have a loose wiring? I soldered them to the board and they seem solidly attached to the pads. However, when I wiggle SWDIO and restart openocd it sometimes briefly works (gets to the dump message). I'm struggling to isolate the wire or figure out a pattern to help me troubleshoot. I didn't find any issues yet with a multimeter continuity test. I guess I'll try desoldering the connection tomorrow.

root@ubuntu:~/airbreak# openocd -f ./tcl/airsense.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
Info : replaced existing 'mrw' usage
Info : replaced existing 'mrw' help
Info : replaced existing 'mrb' usage
Info : replaced existing 'mrb' help
Info : replaced existing 'mmw' usage
Info : replaced existing 'mmw' help
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 2002 kHz
Info : SWD DPIDR 0x2ba01477
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
This device is not running modified firmware.
File stm32.bin not found!
Backup of stock firmware image not found.
Please type 'dump' to save the stock firmware to disk.

Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 100ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 300ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 700ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 1500ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 3100ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 6300ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 6300ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 6300ms
Polling target stm32f4x.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 6300ms
^C

dblunk88, 2020-04-30:
Just a FYI: I'm running into the same checksum mismatch and I have the same checksum as you

dblunk88, 2020-04-30:
Hash is equivalent to SW version: SX567-0306

nbritton, 2020-04-30:
Good news, it turns out it was a bad connection on the SWDIO wire, I desoldered the old wire and cleaned up the pad and soldered on an entirely new wire. Now openocd is working as expected on the very first attempt. The old wire didn't fail continuity testing, but I think the problem was the wire just had very high resistance relatively speaking that was degrading the digital signaling... the pad is so small and maybe the rosin core I used was interfering with the little bit of contact surface... I don't know, but openocd is working now!

Now I just need to figure out how to injects the rights bits in the right places with the v306 firmware that I have, but ideally I would like to figure out how to upgrade my machine to v401.

dblunk88, 2020-04-30:
Is there a reason why we aren't sharing the firmware dumps? This would make it a lot easier for people who haven't been getting OTA updates

colinoflynn, 2020-04-30:
The firmware is copyright material, and should **never** be shared (or even links to such material submitted). People doing so would jeopardize the entire project.

Note that the same SW version number encompasses a large group of potentially incompatible firmware as as well. The firmware (and location of stuff) changes also with the 'catalog no' on the back. It's possible some catalog no's are never OTA to the latest FW version as well, which could be due to compatibility reasons for example.

dblunk88, 2020-04-30:
Aw, that makes sense! Thank you for the clarification and for all of the hard work you and others have put in to this project.

nbritton, 2020-04-30:
This is from the patch-airsense script:

extra_modes() {
        BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 3) ))
        # add more mode entries, set config 0x0 mask to all bits high
        # default is 0x3, which only enables mode 1 (CPAP) and 2 (AutoSet)
        # ---> This is the real magic <---
        printf '\xff\xff' | patch 0x8590 || die failed
}

From someone who has a v401 firmware, what exactly is located at 0x8590? I'm wondering if I can use xxd and just grep for that string to find the correct location in my v306 firmware to patch. A grep for just 0303 returns too many entries.

gszakacs, 2020-04-30:
> Is there a reason why we aren't sharing the firmware dumps? This would make it a lot easier for people who haven't been getting OTA updates

It does not appear that the manufacturer is pushing updates to devices, It seems that patients need to go to their doctor to have the devices updated to newer firmware versions. 

Ref: http://www.apneaboard.com/forums/Thread-Resmed-Airsense-10-Autoset-firmware-update?page=3

dblunk88, 2020-04-30:
> This is from the patch-airsense script:
>
> extra_modes() {
> BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 3) ))
> # add more mode entries, set config 0x0 mask to all bits high
> # default is 0x3, which only enables [...]

I've asked around and may have found a friend with the newer firmware who would let me borrow his device. Since I'd have access to both dumps, I could see if I can compare both

nbritton, 2020-04-30:
I have new modes now after my latest flash:

[image]

[image]

[image]

root@ubuntu:~/airbreak# git diff patch-airsense
diff --git a/patch-airsense b/patch-airsense
index b0e58d4..4b315e2 100755
--- a/patch-airsense
+++ b/patch-airsense
@@ -20,7 +20,7 @@ patch() {
        dd bs=1 seek=$offset conv=notrunc of="$OUT" status=none
 }

-echo "533b91127aa22e05b933db203ad56c449dc12a8c3fd62f57bd88c472a8061775  $IN"\
+echo "363a204ba217f31223e929365d58b8f5ce038a7681e362fe157e190c2eacbd30  $IN"\
 | sha256sum --check \
 || die "$IN: wrong hash"

@@ -32,8 +32,8 @@ printf '\xc0\x46' | patch 0xf0 \

 # and add a message so that we know this is a modified firmware
 printf 'HACKED!' | patch 0x17500 || die failed
-printf 'NOT FOR USE\x0' | patch 0x1a540 || die failed
-printf 'WARNING! WARNING! Ventilator test firmware: Not for humans!\x00' | patch 0x1b860 || die failed
+#printf 'NOT FOR USE\x0' | patch 0x1a540 || die failed
+#printf 'WARNING! WARNING! Ventilator test firmware: Not for humans!\x00' | patch 0x1b860 || die failed

 BUILD_FLAGS=0

@@ -115,12 +115,12 @@ gui_config () {
        done
 }

-patch_code
-unlock_ui_limits
-extra_debug
+#patch_code
+#unlock_ui_limits
+#extra_debug
 extra_modes
 extra_menu
-#all_menu
+all_menu
 gui_config

 FLAGSTR=$(printf 'FLAGS=0x%02x' $BUILD_FLAGS)

dblunk88, 2020-04-30:
> I have new modes now after my latest flash:

Oh nice! Is everything working?

nbritton, 2020-04-30:
Still playing with it, not even sure what half the features do! However, the first one I tried, ASVAuto, appears to be working... because if I pause my breath for a few seconds the machine will increase the pressure to force me to inhale.

The screen has some non-sense on it too...

[image]

dblunk88, 2020-04-30:
iVaps has been acting strange for me, just wondering if it is the same for you

nbritton, 2020-05-01:
I'm not even sure what that is so I'm probably not the best person to ask if it's operating normally. I'll do what I can to help out converting these to emergency ventilators, but I came here primarily for enabling BiPAP / ASVAuto functionality on my own personal machine. I'm a computer systems engineer who also likes to do electrical engineering as a hobby, so I was compelled to pull mine apart and tinker with it. :-)

[image]

dblunk88, 2020-05-01:
Would you still like the dump comparisons?

nbritton, 2020-05-01:
Yes absolutely, I think that is critical for enabling emergency ventilator use because it seems very apparent that ResMed has a multitude of firmware versions in production units. If we can figure out how to find the right insertion points in the code for the 0x0 mode feature mask we can automate the process to work on most any arbitrary firmware version.

nbritton, 2020-05-01:
I think the very first time I re-flashed the machine it was a bad upload, recall that my IO wire was flakey... the extra menus are working now with just the 0x0 mode feature mask. This code works as is verbatim with my firmware version:

```
extra_modes() {
        BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 3) ))
        # add more mode entries, set config 0x0 mask to all bits high
        # default is 0x3, which only enables mode 1 (CPAP) and 2 (AutoSet)
        # ---> This is the real magic <---
        printf '\xff\xff' | patch 0x8590 || die failed
}
```

dblunk88, 2020-05-01:
Left is 0401 offset, right is the equivalent 0306 offset using a hexdump of both bins. Basically just went through the script and fished out all relevant offsets.
00008590  --> 00008590
00017500 --> 00017500
0001a540 --> 0001a540
0001b860 --> 0001b860
**000bb730 --> 000bb4d0**
000fd000 is an unwritten offset
000f9c80 --> 000f9a20
0004fa80 --> 0004fa80
00004fc0 --> 00004fc0
00007eb0 --> 00007eb0
00007ee0 --> 00007ee0
00007ec0 --> 00007ec0
000084a0 --> 000084a0
00008590 --> 00008590
00066470 --> 00066470 Note: 0006647c and 0006647d are different values
0006e500 --> 0006e500
0006e4c0 --> 0006e4c0
00004ef0 --> 00004ef0
00017580 --> 00017580
00017760 --> 00017760

As can be seen, there are not a lot of differences

dblunk88, 2020-05-01:
Added a pull request based off of what I found to make patch-airsense compatible

gszakacs, 2020-05-01:
Patched a v306 bin file using dblunk88's update, but I do not see new modes either in the user or the clinical menu.

dblunk88, 2020-05-01:
@gszakacs comment out patch_code

Like so:
`#!/bin/bash

die() { echo >&2 "$*" ; exit 1 ; }	

IN=$1
OUT=$2

if [ -z "$IN" -o -z "$OUT" ]; then
		die "Usage: $0: stm32.bin stm32-unlocked.bin"
fi

if [ ! -r "$IN" ]; then
		die "$IN: can not open input file"
fi

patch() {
	offset=`printf "%d" $1`
	printf "patching %08x\n" $1
	dd bs=1 seek=$offset conv=notrunc of="$OUT" status=none
}

shasum=$( sha256sum $IN | cut -f 1 -d " " )
if [ $shasum = "533b91127aa22e05b933db203ad56c449dc12a8c3fd62f57bd88c472a8061775" ]; then
	version="401"
elif [ $shasum = "363a204ba217f31223e929365d58b8f5ce038a7681e362fe157e190c2eacbd30" ]; then
	version="306"
else
	die "$IN: incompatible firmware version."
fi
echo "Found firmware version: $version"

cp "$IN" "$OUT" || die "$OUT: copy failed"

# also patch up the jump instruction that checks for tamper
printf '\xc0\x46' | patch 0xf0 \
|| die "startup patch failed" \

# and add a message so that we know this is a modified firmware
printf 'HACKED!' | patch 0x17500 || die failed
printf 'NOT FOR USE\x0' | patch 0x1a540 || die failed
printf 'WARNING! WARNING! Ventilator test firmware: Not for humans!\x00' | patch 0x1b860 || die failed

BUILD_FLAGS=0

patch_code() {
	#if you want to add the extra breath mode
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 0) ))
	if [ -r breath.bin ] && [ $version = "401" ] ; then
		echo "Including breath.bin"
		cat breath.bin | patch 0xbb734 || die "binary patch failed"
	elif [ -r breath.bin ] && [ $version = "306" ] ; then
                echo "Including breath.bin"
                cat breath.bin | patch 0xbb4d4 || die "binary patch failed"
	else
		die "breath.bin not found, did you run make?"
	fi 

	if [ -r graph.bin ]; then
		echo "Including graph.bin"
		cat graph.bin | patch 0xfd000 || die "graph patch failed"
		printf '\x01\xd0\x0f\x08' | patch 0xf9c88 || die "graph fixup failed"
	fi
}

unlock_ui_limits() {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 1) ))
	# patch UI limits to range between 1 and 30
	printf '\xdc\x05\x00\x00\x32\x00' | patch 0x4fa8 || die failed
	printf '\xdc\x05\x00\x00\x32\x00' | patch 0x4fc4 || die failed
	printf '\xdc\x05\x00\x00\x32\x00' | patch 0x7eb0 || die failed
	printf '\xdc\x05\x00\x00\x32\x00' | patch 0x7ee8 || die failed
	printf '\xdc\x05\x00\x00\x32\x00' | patch 0x7ecc || die failed
}

extra_debug() {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 2) ))
	# set config variable 0xc value to 4 == enable more debugging data on display
	# if you set it to \x0f it will enable four separate display pages of info in sleep report mode
	printf '\x04' | patch 0x84a8 || die failed
}

extra_modes() {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 3) ))
	# add more mode entries, set config 0x0 mask to all bits high
	# default is 0x3, which only enables mode 1 (CPAP) and 2 (AutoSet)
	# ---> This is the real magic <---
	printf '\xff\xff' | patch 0x8590 || die failed
}

extra_menu() {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 4) ))
	# try enabling extra menu items
	printf '\x01\x20' | patch 0x66470
}

# If you want all menu items to always be visible, let this section run
all_menu() {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 5) ))
	# force status bit 5 always on -- always editable
	printf '\x01\x20' | patch 0x6e502
	# force status bit 4 always on -- this makes all the inputs show up, regardless of mode
	printf '\x01\x20' | patch 0x6e4c4
}

gui_config () {
	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 6) ))
	# enable all of the editable options in the settings menu
	# by turning on bit 1 of the config entries.  All of these variables
	# are listed in the gui_create_menus function
	GUI_CONFIG=0x4ef4
	GUI_CONFIG_SIZE=0x1c
	GUI_CONFIG_OFFSET=30

	for var in \
		0x2f 0x1ec 0x1ed 0x24 0x25 0x1d3 0x1d6 0x1d5 0x1d7 0x26 \
		0x1d9 0x1e0 0x1e1 0x1e2 0x1e5 0x1e4 0x1e6 0x1e7 0x1e9 0x1ea 0x1eb \
	; do
		# set config byte 0 bit 0, 1 and 2, which makes it active
		ADDR=`perl -e "printf '%d', $GUI_CONFIG + ($var - $GUI_CONFIG_OFFSET) * $GUI_CONFIG_SIZE"`
		printf '\x07\x00' | patch $ADDR
	done
}

#patch_code
unlock_ui_limits
extra_debug
extra_modes
extra_menu
all_menu
gui_config

FLAGSTR=$(printf 'FLAGS=0x%02x' $BUILD_FLAGS)
printf $FLAGSTR | patch 0x17588

COMMIT_HASH=$(git log -n1 --format=format:"%H" | head -c 7)
printf 'GIT=%s\x00' $COMMIT_HASH | patch 0x17764

sha256sum $OUT`

dblunk88, 2020-05-01:
Also, I will probably add an automatic hexdump search to find offsets sometime in the future to get all versions compatible (if there are any non-306 and 401 versions floating around)

gszakacs, 2020-05-01:
@dblunk88 You are correct. I used the patch file from your repo and run it on my bin,
After following your most recent comments, more items were patched, but I still received  a git related fatal error between items 17588 and 17764, Not sure how critical that is. 

After patching, I also have the same non-sense menu titles as @nbritton + noticed duplicate "min PS", "max PS", "start EPAP" items in the clinical menu. Some settings are visible but not adjustable, for example the "rise time" "trigger" "cycle". As far as the emergency ventilator functionality those menu items are hopefully irrelevant, but I am curious if you are noticing the same limitations.

dblunk88, 2020-05-01:
I think the menu items are intentional and may be used as an indicator on what version it is running. You can actually see them being patched in at the end of the script.

As far as the limited options go, I do have the same issue, but really haven't taken the time yet to see on why that is. I'm still getting familiar with the binaries myself... someone else might know the answer though :)

Could you paste the error?

ruri-baka, 2020-05-02:
I've been working on 0302 compatibility and here is what I have so far:

unlock ui limits, extra debug, extra modes, and the gui config address need a +8 offset

comment out extra menu and the second address of all menu, then offset the first (status bit 5) address by +8

The anti-tamper may not be needed but it does need to be offset by +2 to use it.

I haven't tried anything else but that is tested and working on a Rev 1.0 board as well. What I've ran into though is that the GUI areas on 302 are a bit different. For example (from the 0401 stubs file):

NSTUB(0x08066c1b, dispatch_0x38_0x40) is at 664f6 -725
NSTUB(0x080668dd, date_format_time) is at 6589a (I think)

I haven't tested anything else yet but that yields working menus, graphics, and all modes; again on SX567-0302 on a Rev. 1.0 PCB.

dblunk88, 2020-05-02:
@ruri-baka awesome!! You should do a pull request :)

I wonder how many 302's are floating out there

ruri-baka, 2020-05-02:
Lots..sitting in warehouses...

By the way...

Text Strings: not for use is at 1a5bo, the other two are the same.

SHA256: A5E7D77FC7B2FE38DA82A7D99035CECD40749398DA31F7E39634C002E7081CC7

dblunk88, 2020-05-02:
Mind sharing your patcher? I could add this to my existing pull request if you want

ruri-baka, 2020-05-02:
I just created a pull request with it; I'd be interested to see if what I did for the menus on 0302 would get them working on 0306 though.

dblunk88, 2020-05-02:
My concern is having two different versions for the patcher. Once either pull gets through, I'll merge them

dblunk88, 2020-05-02:
Also, since I have access to a 306 and temporary access to a 401 device, I can do some testing for you if you want

ruri-baka, 2020-05-02:
Go for it; I didn't know who's working on what patcher so I figured I'd throw that up so I can take a break and start digging and debugging the additions (graphs, debug, etc.) for 302.

If you have two versions you might want to run some diffs on the 6xxxx space...are they very similar between 306-401?

dblunk88, 2020-05-02:
The python patcher is working really well, might just add on to that instead

dblunk88, 2020-05-02:
@ruri-baka Will take a look :) Do you think we should open up a different issues for the comparisons? This thread is starting to feel a bit cluttered

dblunk88, 2020-05-02:
Scripting out a python program to grab the individual offset differences between binary files (through the use of xxd -c 1 <file>) that'll split out an excel file

dblunk88, 2020-05-03:
Done! Put in another pull request. https://github.com/dblunk88/airbreak/blob/master/offset_diff.py

Will upload a redacted version of the results on my Drive.

dblunk88, 2020-05-03:
Here are the results
https://docs.google.com/spreadsheets/d/1_i6PF-nABOaD_m5coIUW_zA4tfrCzPFv0MkYMmZaUq8/edit?usp=sharing

ruri-baka, 2020-05-03:
Well that explains why I was going nuts over the trying to get the GUI to work, but the good news is that you should be able to enable the menus on 306 with the 302 menu code. 

Do all the menus show up in 401 (for example height and back up rate in ivaps)?

> @ruri-baka Will take a look :) Do you think we should open up a different issues for the comparisons? This thread is starting to feel a bit cluttered

Yeah...This is gonna get confusing real fast in one issue

dblunk88, 2020-05-03:
I wish I had 302 as a reference. I'll ask around to see if I can find someone and borrow a device running that specific firmware. :) I think most, if not all, menus show up. But can't edit some of them and there are a few duplicates.

dblunk88, 2020-05-03:
Also will be adding a offset finder soon to see where changed offsets might be located at for differing firmware versions.

dblunk88, 2020-05-03:
Above is done. Added a new issue at https://github.com/osresearch/airbreak/issues/41

nimamoslemy, 2020-05-03:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)

nbritton, 2020-05-04:
I was able to disassemble the binary using:

```
arm-none-eabi-objdump -D -bbinary -marm stm32.bin -Mforce-thumb > stm32.s
```

Additionally, The Reko program was able to disassemble it as well.

https://github.com/uxmal/reko

The main chip on my AirSense 10 AutoSet is STMicroelectronics STM32F405ZGT6. This is an ARM Cortex-M4 32-bit RISC, however the Internet says the Cortex-M series only supports the 16-bit Thumb ISA.

STM32F405ZG Datasheet:
https://www.st.com/resource/en/datasheet/stm32f405zg.pdf

16-bit Thumb ISA Quick Reference:
http://infocenter.arm.com/help/topic/com.arm.doc.qrc0006e/QRC0006_UAL16.pdf

[image]

ruri-baka, 2020-05-04:
> I was able to disassemble the binary using:
>
> ```
> arm-none-eabi-objdump -D -bbinary -marm stm32.bin -Mforce-thumb > stm32.s
> ```
>
> Additionally, The Reko program was able to disassemble it as well. [...]

Here is a link to STMicro’s STemWin package; it includes libraries and headers for the STM32F405ZG and additional IO as well as emwin headers and libraries for that processor:  https://www.st.com/en/embedded-software/stemwin.html#overview

Asmageddon, 2021-06-20:
Hey @nbritton, @ruri-baka, @dblunk88, and/or @colinoflynn, could I solicit any of you for help/assistance/advice with working with airbreak and openocd that goes slightly past the scope(not in complexity) of this project and issues relating to it?

nbritton, 2021-06-20:
> Hey @nbritton, @ruri-baka, @dblunk88, and/or @colinoflynn, could I solicit any of you for help/assistance/advice with working with airbreak and openocd that goes slightly past the scope(not in complexity) of this project and issues relating to it?

Well I've been reassigned to a new project now that there isn't a ventilator shortage due to the pandemic, so I'm not sure how much assistance I can offer you, but what would you happen to be having trouble with?

Asmageddon, 2021-06-20:
@nbritton Basically, I've been trying to flash firmware from one machine(bought a broken ASV) to another, but am finding that it is failing to boot(the airbroken ASV mode does not work correctly), with the main three possibilities I can think of being that it won't boot older firmware(0306 vs my 0401), that there might be some sort of region restriction(if that's even a thing), or the fact that optionbytes differ. I tried patching the tamper check byte, but no dice, and I'm wary of fiddling with optionbytes in case it results in a brick. If possible, I would like to diagnose and overcome the problem, as regrettably, my VAuto is simply not cutting it for me, and I'm out of capacity to afford further/alternate treatment for the time being.

nbritton, 2021-06-20:
My educated guess is that ResMed "fixed" their firmware to prevent this, as producing a single physical machine that has identical hardware in it and then charging insurance carriers different rates to "enable" software features would probably be considered insurance fraud. Especially, given the fact that they force insurance carriers to purchase an entirely new device for the patient instead of just allowing the DME supplier to flip a few bits in software to upgrade the device in the future and bill the difference.

Honestly, what we really need at this point is an open source flow generator that has an open standards API, as the precision movement of air has many industrial and scientific applications beyond helping people breath. If ResMed is smart, they should produce this open device or at least the specifications for it, and then ask the FDA to approve their software defined breathing algorithm to run on the device. The money isn't in the hardware, outsource that part, what is valuable is the software defined breathing algorithm combined with FDA approval.

nbritton, 2021-06-21:
In fact, I read an article last week that Apple was trying (and failing) to make a major push into the Health industry... ResMed, Respironics, and Fisher & Paykel should team up with Apple to produce a "Software Defined Flow Generator". ResMed, Respironics, and Fisher & Paykel can provide their expertise as far as the flow generator, accessories, and consumables are concerned, and Apple can provide their expertise in software and embedded SoC platforms using their A series ARM processors. Throw in some extra sensors for telemetry (oximetry and knowing if the patient is asleep or awake would be nice), provide Apple Watch, iOS HealthKit, and Hey Siri integration, and create an App Store for ResMed, Respironics, and Fisher & Paykel to publish their FDA approved Software Defined Breathing Algorithms to. Provide a remote access framework so DMEs, Physicians, and Insurance carriers can interact with the device. Insurance carriers would probably love this, as they can just pay a monthly subscription for the Breathing Algorithm "App" and if the patient is not compliant they can simply stop pay for the subscription, or for instance if the doctor changes their prescription, i.g. traditional CPAP, auto CPAP, fixed BiPAP, auto BiPAP, ASV, ST, the DME can change the settings.

Then when the next respiratory pandemic hits we will have several million Software Defined Flow Generators stockpiled that can be reprogramed into ventilators in just a few seconds.

https://www.youtube.com/watch?v=K84ymNFuu6U

nbritton, 2021-06-21:
With regard to reprogramming the Software Defined Flow Generators into ventilators, you wouldn't even need the infected patients to come to a hospital, you can send EMS to their home and have a video teleconference with a physician to set it up, and with telemetry the physicians can monitor the patients remotely and a member of the family can coordinate care. The next variant might be much more deadly.

Asmageddon, 2021-06-21:
I know for a fact that relatively recent machines can still be flashed with alternative firmware, I know a person who has done it. On my own, the jailbreak also works, and it boots again when flashed with the original firmware. I've got good reason to believe that the issue is a plausibly debug-able one, but doing so exceeds both my expertise and my cognitive energy reserves.

That's an interesting concept really, and given how profitable subscription-based models are, it just about might make it.
Ideally, we'd want freely available OSHW solutions that come with fine-tuned ASV software - CPAP has no reason to exist, APAP is "lazy doctor CPAP", BiPAP is woefully insufficient to meet variable respiratory demand without overventilating, and exist only because product stratification at the cost of the health of patients is profitable.

As much as I'm sure that the next ventilation-demanding pandemic will indeed happen, I personally care about all the under- or un-treated Sleep Disordered Breathing patients - only 5 to 20% of apnea patients are diagnosed depending on region, vast majority are undertreated on basic CPAP or APAP, and Upper Airway Resistance Syndrome patients, who make up the majority of the spectrum between healthy sleep breathing and severe airway collapse, are almost never diagnosed, and rarely treated properly- no amount of airway-stenting pressure is enough to prevent escalation of respiratory effort against resistance imposed by physically narrow airways. We're talking about numbers all the way to possibly a third of humanity who could use an ASV on their way to ultimate resolution via orthognatic surgery.

Once I have my own condition under control(wish I could say it will be soon, but I don't think I'm making it short of orthognatic surgery), I very much so intend to attempt to design such an OSHW device- Software Defined Flow Generator being a concept as catchy as any- and try to get it through regulation(or bypass it altogether), to try to let people get the lives that insomnia, nightmares, fatigue, REM dream deprivation, mental and physical health issues caused by SDB, are taking from them, without having to spend thousands of dollars.

I do not trust the corporate to do anything but try to beggar people and cripple their software to create demand for alternative/future products - as they do with product stratification, with ResMed devices not featuring exhale support, with price gouged supplies.

Asmageddon, 2021-08-14:
@nbritton Ping.

Biorn1950, 2021-08-14:
>CPAP has no reason to exist, APAP is "lazy doctor CPAP"
>
Nonsense.
> vast majority are undertreated on basic CPAP or APAP
> 
How you can affirme that? Where is your study repport.

Asmageddon, 2021-08-14:
@Biorn1950 Static pressure stents airways and prevents collapse, but it also increases expiratory effort, which prevents it from being able to resolve Respiratory Effort Related Arousals the way it can resolve Apneas/Hypopneas. Over 60% of people can not tolerate CPAP, and the reason is not "just discomfort"- people who benefit from PAP therapy, tolerate it. 

APAP increases pressures in response to events and to severe flow limitations - after they happen, which is just not good enough. I work with a lot of people who have been prescribed APAP in our online community, and very often, it is left on default settings, such as 6-12 or 6-20, with very wild swings of pressure throughout the night, with people experiencing disruption as the pressure drops again. It has actual - if minimal - benefit over CPAP if the baseline pressure is titrated to where CPAP pressure would be thanks to being able to increase EPAP during REM sleep, but the problem from the prior paragraph still applies.

CPAP/APAP can work for some people, especially older people, but that requires the condition to either be mild enough to not run into Expiratory Pressure Intolerance, or for them to have damage from years of Sleep Disordered Breathing and snoring that prevents RERAs, instead promoting apneas, in which case just stenting the airway is enough.

I'm trying to focus on my own condition, and dial back on expending my effort on helping others while my own condition is still very bad, so you will have to look for sources yourself. The one link I have handy is Barry Krakow's clinical anecdote, that I believe he might have published research about, or is going to soon: https://www.sleepreviewmag.com/sleep-disorders/breathing-disorders/a-missing-link-dr-barry-krakow-s-research-on-insomnia-and-sdb/

Biorn1950, 2021-08-14:
> Over 60% of people can not tolerate CPAP
> 
You just invent numbers.
And expiratory effort don't exist due to the fact no muscle intervenes during the exhalation.
About APAP I don't know where you live but in my country it's the doctor who tell to the technician to adjust parameter after the first month of usage. Not the patient.

Asmageddon, 2021-08-14:
Exhalation has a passive phase, and a late active exhalation phase. Furthermore, the airway is not static during exhalation and exhalatory resistance contributes to the process that integrates respiratory effort to culminate in an arousal.
And yeah, that's how it should look, but there's two problems with this- without SD card access, over the air data only gives limited information that is not sufficient for full judgment of how to adjust parameters, and second: it's common for the doctor to just not do this and leave patients on initial settings for months. Refer to the following writeup, also by Krakow: https://www.ncbi.nlm.nih.gov/labs/pmc/articles/PMC6933255/

[30 more replies]