osresearch/airbreak #37: Raspberry Pi HowTo

nbritton, 2020-04-30:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

**Ubuntu 20.04 Raspberry Pi 3 Image Download Link:** https://ubuntu.com/download/raspberry-pi/thank-you?version=20.04&architecture=armhf+raspi

[image]

[image]

[image]

[image]

[image]

gszakacs, 2020-04-30:
Thanks to your pics, I was able to troubleshoot the issue with my hardware setup and able to dump the bin file.

dblunk88, 2020-04-30:
Is there any chance that this can be added to the wiki?

davelatt, 2020-06-11:
Hi Nikolas,

My name is Dave Lattimore and I saw you Post on the apnea boards about enabling ASV. I had the thread bookmarked but they removed the post. Guess someone didn't like the idea of an AirBreak :-). I was going to private message you to see if there are any issues with missing ASV data structures but I don't have you logon id. My "apnea board" and "cpaptalk" id is sosotired. My email is davelatt@hotmail.com.  Please get in touch.
Thanks,
Dave

amorton12, 2022-05-11:
For any future hackers that come across this and are using a RasPi B version one, like me, you’ll need to do the following:

Wire your serial clock line to the SPI0 sclk pin GPIO11, data to GPIO25, reset to GPIO24. 

Modify /usr/share/openocd/scripts/interface/raspberrypi-native.cfg 

Uncomment the two lines 
bcm2835gpio_srst_num 24
reset_config srst_only srst_push_pull

Modify the rasp-pi.cfg to source the raspberrypi-native.cfg rather than raspberrypi2-native.cfg

The rest works the same. I successfully dumped and flashed the 0306 image from both a cpap and an autoset version of the Airsense 10 with this config (and verified that they are interchangeable).

jayhathaway, 2022-08-28:
> For any future hackers that come across this and are using a RasPi B version one, like me, you’ll need to do the following:
>
> Wire your serial clock line to the SPI0 sclk pin GPIO11, data to GPIO25, [...]

Cool thanks! So did your flash with 0306 go smoothly without any issues?

ttkirst, 2022-11-20:
> > Cool thanks! So did your flash with 0306 go smoothly without any issues?
> 
> Yes, I flashed the Autoset stock image and the unlocked/modified image onto the base model CPAP machine with this. Both worked perfectly, and I've been using that machine daily in Autoset mode for about 4 months.

Hi, I am trying to airbreak my Airsense 10 CPAP elite with a raspberry pi 4 as the programmer. I follower your instructions but I cant seem to get it to work properly. It connects to the uC but won't dump the firmware and keeps giving errors. Could you please assist?

Kind regards,
Tyler

root@raspberrypi:/home/pi/airbreak# sudo openocd -f tcl/rasp-pi.cfg -f tcl/airsense.cfg
Open On-Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
0x2000
Info : replaced existing 'mrw' usage
Info : replaced existing 'mrw' help
Info : replaced existing 'mrh' usage
Info : replaced existing 'mrh' help
Info : replaced existing 'mrb' usage
Info : replaced existing 'mrb' help
Info : replaced existing 'mmw' usage
Info : replaced existing 'mmw' help
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 2001 kHz
Info : SWD DPIDR 0x2ba01477
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
This device is not running modified firmware.
File stm32.bin not found!
Backup of stock firmware image not found.
Please type 'dump' to save the stock firmware to disk.

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 100ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 300ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 100ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 300ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 700ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 1500ms
Info : SWD DPIDR 0x2ba01477
Polling target stm32f4x.cpu failed, trying to reexamine
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Polling target stm32f4x.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 100ms

ttkirst, 2022-11-20:
This is the error I get when I try dump the firmware... Please someone assist, been trying for 2 days straight...

root@raspberrypi:/home/pi/airbreak# telnet localhost 4444
Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> reset halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
> dump
Dumping stm32.bin, 0x100000 bytes at 0x08000000...
SWD DPIDR 0x2ba01477
Failed to read memory and, additionally, failed to find out where

amorton12, 2022-11-20:
> 

> Hi, I am trying to airbreak my Airsense 10 CPAP elite with a raspberry pi 4 as the programmer. I follower your instructions but I cant seem to get it to work properly. It connects to the uC but [...]

Disclaimer: I am far, far from any kind of expert here. This was my first time trying to talk to any kind of STM microcontroller 

You definitely don’t want to follow my instructions with a Pi 4, but you may need to do what I did. I looked up the port/pin assignments for my version one Pi compared to the instructions that were written for a Pi 2/3, and made those adjustments so that it was using the same port functions, just on different pins. 

I would guess that the original instructions are probably closer to what you need than mine, since mine are adapted for a much earlier revision of the Pi hardware. It took me a few days of hacking at it and taking breaks before I figured it out, so you’re not the only one to struggle with it. I had actually ordered a USB programmer and it was in the mail when I finally figured it out, so worst case scenario you can throw in the towel and spend an extra $10 or so to get it done.

ttkirst, 2022-11-20:
> Disclaimer: I am far, far from any kind of expert here. This was my first time trying to talk to any kind of STM microcontroller
>
> You definitely don’t want to follow my instructions with a Pi 4, but [...]

Thanks so much for the reply, I really appreciate it.
I did basically follow the original, just made the changes to suite pi 4 as you suggested. Think the hardware is basically working since I can connect and dump some data. My issue is it won't dump all. The reason I came to your post is mainly due to the firmware difference, thought maybe that had a part to play. Mine is 0302... Was trying to figure out how you managed to patch it? I saw the attached patch file above but not sure how to use it or what file to apply the patch to? Does it replace the current airsense patch file? and use it to patch my firmware when I manage to dump it? Or...?

On a different note, this is what I get when I use dump_all, seems to fail at the same place...

> dump_all
Dumping syscfg.bin, 0x400 bytes at 0x40013800...
Dumping bkpsram.bin, 0x1000 bytes at 0x40024000...
Dumping rtc_bkpr.bin, 0x50 bytes at 0x40002850...
Dumping optionbytes.bin, 0x8 bytes at 0x1fffc000...
Dumping bank0.bin, 0x100000 bytes at 0x08000000...
SWD DPIDR 0x2ba01477
Failed to read memory and, additionally, failed to find out where

Thanks so much for your feedback, good to know its not just me struggling.

amorton12, 2022-11-20:
The dump process will be the same regardless of version - the difference comes in with the actual patching, since the key pieces of data that need to be flipped around will be at different offsets in the file - I think there’s a patch or a fork out there with a file that contains the correct offsets for your firmware, but don’t quote me on that - it’s been quite a while since I worked on this. 

I do have a Pi 400, which afaik, has the same pin out as the Pi 4, so if I have a need to take one of my machines apart, I may fire it up and take a crack at it. If I have any luck, I’ll post here again with whatever worked.

ttkirst, 2022-11-20:
> The dump process will be the same regardless of version - the difference comes in with the actual patching, since the key pieces of data that need to be flipped around will be at different offsets [...]

Cool, thanks so much. If I manage to get it working I will add details to this thread to hopefully help the next person. At the moment I’m thinking it is protection bits or something along those lines preventing me from reading that location. The error “could not find MEM-AP to control the core” is also a little concerning 🙈