osresearch/airbreak #38: Compiling and bugs

mpiani, 2020-05-01:
I'm sorry to ask a very newbie question, but given the relevance of the project would someone care to give some hints on how to compile the project? I was able to extract the firmware. Although the firmware version was the same, the checksum was different, fixed that, the compiler asks for breath.Bin and asks if I performed make, which I did without result. Another problem is stubs which says there is no target. How should I proceed, please?

dblunk88, 2020-05-01:
Technically (as far as I understand it) you do not need to make, unless you want breath and graph. For patch-airsense, just comment out patch_code or reference https://github.com/osresearch/airbreak/issues/33

mpiani, 2020-05-01:
Thank you for the reply. I was able to generate unlocked-stm32.bin by adjusting the hash on the patch, but my understanding is that without breath.bin the cvap and other  functions would not become available. Can anyone confirm that? If so, how to compile breath.c or breath.o and the others?

dblunk88, 2020-05-01:
I'm not sure, but I think the compilation requires a xml generated by Ghidra (could try importing it using cortex as the language). Haven't really looked into it too deeply this far. Nonetheless, if you do compile it and generate the firmware, be aware the hex comparisons revealed that the breath.bin offset location for 306 is different than the 401 one for breath.bin, but it may still run (like said, haven't taken the time yet to test this yet).

gszakacs, 2020-05-02:
@dblunk88 is there a way to PM you ?

dblunk88, 2020-05-02:
I don't know if I'd be the best person to PM, but sure. I could temporarily post my discord, Twitter, or E-Mail If you want.

gszakacs, 2020-05-02:
you seem to be the most active/helpful. I don't really need help just have a question about 401. email would be great. thanks.

dblunk88, 2020-05-02:
@gszakacs sure: david@blunk.cc

Let me know when you have it