osresearch/airbreak #41: Compatibility across all firmware versions

dblunk88, 2020-05-03:
Here is a direct comparisons between the 401 firmware and the 306 firmware on where relevant offsets might be located at:
https://docs.google.com/spreadsheets/d/1VjTxSbz4m1rdDFkt8cg_n5N7Q3xiDmSVyzu5a7N4I08/edit?usp=sharing

About to do a pull request to share the code. The original can be found on my fork: 
https://github.com/dblunk88/airbreak/blob/master/offset_locator.py

Usage:
First do a xxd on the 401 firmware and whatever firmware you are trying to compare it with, like so:
"xxd <file> > <file>.hex"
Then run the program as so:
"python list_offset_diff.py <hexfile1> <hexfile2>"
Since this Github is primarily focused on programming for 401, hexfile1 should be the xxd of the 401 firmware

Output will be a CSV which can be imported easily into Excel (to easily CTRL+F).
It only lists the offset if they are different (so if the offset isn't listed, it is implied to be the same).
There may be multiple matching offsets, since this is a list of ALL matching offsets. If there is no match, then that does not mean that it is completely lost, it is just not found in sectors split by 0-F's and a tighter search may be the recommended action.

This should make compatibility easier between all versions :)
Note: It's not very efficient. Bring some time and coffee. If there is a high demand (doubt it, since this technically could be run once and it should mostly suffice), then I could see if I can build in a SQLite search via query (which should speed things up a LOT).

nbritton, 2020-05-05:
I have an AirCurve 10 S on the way so I should be able to help with getting the differential. Not sure what version of the firmware it has on it, but I'm hoping it's the same version as the one on my AirSense 10 AutoSet, because then I think that would help me diff through the parameter data in the firmware for differences.

dblunk88, 2020-05-06:
@nbritton that would be awesome! Let me know and I'll add it to my Google Sheets

nbritton, 2020-05-09:
```
% diff -U0 stm32.bin.AirSense_10_AutoSet_0306.DAT_08008584 stm32.bin.AirCurve_S_0306.DAT_08008584
--- stm32.bin.AirSense_10_AutoSet_0306.DAT_08008584	2020-05-09 14:39:28.000000000 -0500
+++ stm32.bin.AirCurve_S_0306.DAT_08008584	2020-05-09 14:37:47.000000000 -0500
@@ -10 +10 @@
-        0800858c 01              ??         01h
+        0800858c 03              ??         03h
@@ -14 +14 @@
-        08008590 03              ??         03h
+        08008590 09              ??         09h
@@ -130 +130 @@
-        08008604 00              ??         00h
+        08008604 01              ??         01h
@@ -154 +154 @@
-        0800861c 03              ??         03h
+        0800861c 07              ??         07h
@@ -202 +202 @@
-        0800864c 06              ??         06h
+        0800864c 07              ??         07h
@@ -210 +210 @@
-        08008654 00              ??         00h
+        08008654 01              ??         01h
@@ -222 +222 @@
-        08008660 06              ??         06h
+        08008660 07              ??         07h
@@ -394 +394 @@
-        0800870c 3f              ??         3Fh    ?
+        0800870c 1f              ??         1Fh
@@ -402 +402 @@
-        08008714 07              ??         07h
+        08008714 06              ??         06h
@@ -422 +422 @@
-        08008728 06              ??         06h
+        08008728 07              ??         07h
@@ -742 +742 @@
-        08008868 07              ??         07h
+        08008868 06              ??         06h
@@ -762 +762 @@
-        0800887c 07              ??         07h
+        0800887c 06              ??         06h
@@ -782 +782 @@
-        08008890 07              ??         07h
+        08008890 06              ??         06h
@@ -1030 +1030 @@
-        08008988 02              ??         02h
+        08008988 01              ??         01h
@@ -1034 +1034 @@
-        0800898c 07              ??         07h
+        0800898c 03              ??         03h
@@ -1102 +1102 @@
-        080089d0 06              ??         06h
+        080089d0 07              ??         07h
@@ -1122 +1122 @@
-        080089e4 06              ??         06h
+        080089e4 07              ??         07h
@@ -1142 +1142 @@
-        080089f8 06              ??         06h
+        080089f8 07              ??         07h
@@ -1162 +1162 @@
-        08008a0c 07              ??         07h
+        08008a0c 06              ??         06h
@@ -1982 +1982 @@
-        08008d40 06              ??         06h
+        08008d40 07              ??         07h
```