osresearch/airbreak #42: missing menu items in some modes

bdarcus, 2020-05-03:
Some of the modes are missing important, even critical, settings.

In particular, ST, T, PAC are missing ability to set a respiratory rate or backup rate. Without that, I'm pretty sure they're unusable.

Also, "learn targets" and "height" are missing from the iVAPs menus.

This is on a vauto.

gszakacs, 2020-05-03:
I noticed that. Either access to / editing of those menu items or perhaps the precompiled breath.bin/graph.bin would be helpful. 
In case folks are wondering what are the settings for the covid19 related ventilation support modes, you may find the following useful.
[image]
[image]
[image]
[image]

bdarcus, 2020-05-03:
> What firmware do you have ? 

-0401.

> Have you tried compiling/re-flashing with the "all_menu" function enabled ? 

No, in part because I wasn't clear what the purpose of that was. It won't help if items are grayed out.

> I agree that editing those variables may be critical for covid19 related ventiltation.

The RR and BR settings are fundamental to those modes; and yeah, likely critical for covid patients.

nbritton, 2020-05-03:
I can confirm bdarcus original post, these items are not present on my AS10v306. I'm not using the all_menu feature.

bdarcus, 2020-05-04:
So it's confirmed across different machines and fw, including the default autoset and -0401.

I assume this is straightforward to fix, but I don't understand how myself.

bdarcus, 2020-05-06:
Someone did try the all_menu option, and while it does disclose the additional necessary menu items noted here, it's not very practical, since it adds all menus, even when not relevant. We don't want that.

So somehow you need to get these specific menu items added to the standard unlocking.

nbritton, 2020-05-09:
On my AirCurve S with firmware 0306 all of the setting parameters are visible in iVAPS mode (without enabling the force all menus modification), with the exception of Height.

I have a xxd hex diff of my AirSense 10 AutoSet and AirCurve S, both have firmware version 0306. 626 memory addresses have different values between the two firmwares. I'm analyzing the differences.

nbritton, 2020-05-10:
I have good news and bad news. The good news is that the decompiled C source code that Ghidra generated are identical, this basically means that the "runtime engine" is entirely intact. Additional good new is that the hardware is also identical as near as I can tell, both my AirSense 10 AutoSet and AirCurve 10 S both have exactly the same mainboard part numbers.

The bad news, the AirCurve 10 S firmware has data structures / header definitions that are not present in the AirSense 10 AutoSet firmware. This is entirely why the missing items in your original post are missing, so if a menu item is missing that means the data structure isn't defined within the firmware. The firmware would have to be patched to add these data structures.

I did flash my AirSense with my AirCurve's firmware, this worked perfectly and all the data structure definitions that the AirCurve had are present and operational. So let me be clear, this issue is entirely a software limitation. If someone had a genuine AirCurve 10 ST-A firmware this device would have all the data structure definitions and flashing it onto virtually any other device, such as an AirSense 10 AutoSet, would enable a device that has 100% functional iVAPS and PAC modes.

The serial number and other vital product information are not stored within the firmware, that information is stored in an EEPROM; there is no uniquely identifiable information contained in the stock stm32.bin firmware file that OpenOCD dumps.

bdarcus, 2020-05-10:
Hmm ... so in those modes, one is stuck with the fixed default rates.

I imagine that might work for some patients, in some circumstances. But
definitely not ideal.

nbritton, 2020-05-10:
I'm not sure, but it might be more than that. The mode algorithm that the runtime engine is using could simply ignore the parameters that aren't defined. If that's true it would be harder to trust that the algorithm will do the right thing because we don't know how the algorithm uses the sensor and statistical data for making decisions about the flow. For instance, my AirSense didn't know how to derived and track Ti and I:E statistics until I flashed it with the firmware from my AirCurve 10 S.

[image]

[image]

bdarcus, 2020-05-10:
All height does is define a variable to calculate the target MV, which one can do manually also. Why I mentioned it's missing, but not critical.

But RR or BU on the other modes is more important, and what I was referring to when mentioning the "defaults" in my last comment. The modes do run at those default settings without the menus for them.

ruri-baka, 2020-05-11:
It ultimately has to do with how the GUI is compiled for certain sets of machines. The actual values though are still present. It should be possible to set the missing variables through the serial console (see the serial to accessory port pcb) as a workaround.

bdarcus, 2020-05-12:
> It should be possible to set the missing variables through the serial console (see the serial to accessory port pcb) as a workaround.

Probably wouldn't be practical, though, in an emergent hospital environment?

zhuhongchao, 2021-06-28:
> I'm not sure, but I think it might be more than that. The mode algorithm that the runtime engine is using could simply ignore the parameters that aren't defined. If that's true it would be harder to trust that the algorithm will do the right thing because we don't know how the algorithm uses the sensor and statistical data for making decisions about the flow generation. For instance, my AirSense didn't know how to derived and track Ti, Ti Min, Ti Max, and I:E statistics until I flashed it with the firmware from my AirCurve 10 S. In the diff comparison of the AirSense and AirCurve firmwares, the only noticeable difference was the AirCurve had these Ti and I:E data structures defined... these structures were absent in the AirSense firmware.
> 
> [GIF of iVAPS operation on AirSense with AirCurve S firmware](https://i.imgur.com/vF7Btrj.mp4)
> 
> [GIF of iVAPS operation on AirSense with AirSense firmware (all menus visible mod enabled)](https://i.imgur.com/6WdtTv1.mp4)
> 
> **Update:** I noticed in the second GIF that "Height" wasn't marked out with a --, so I tried changing the Height setting and it worked... this then implies that all the necessary data structures are present on the AirCurve 10 S to support iVAPS mode.

I have modified the AIRSENSE AUTOSET firmware according to the instructions and added some new modes, but some parameters are missing. Is it possible to give me the firmware for aircurve?

ArthurMafra, 2022-03-13:
> Some of the modes are missing important, even critical, settings.
> 
> In particular, ST, T, PAC are missing the ability to set a `"Resp. Rate"` or `"Backup Rate"`.
> 
> Without those, one can't really use those modes, which are likely the most useful ones for COVID-19 patients.
> 
> Also, `"Learn Targets"` and `"Height"` are missing from the iVAPS menus.

Hi @bdarcus, may you have Aircurve firmware for share? I have donated device that is buddy only asv auto working. I would like to flash original OEM Aircurve firmware in. If you have backup oem and are willing to help please send to arthuriphone @ gmail. com. Thank you.

jayhathaway, 2023-03-10:
Does anyone have the aircurve vauto firmware they are willing to share privately?

shaareable, 2023-03-10:
@jayhathaway I have a new vauto. You'd need to walk me through it maybe?

DocFreeman, 2023-06-25:
> I actually just got my hands on the firmware. Thank you for offering! On Fri, Mar 10, 2023 at 9:19 AM Jay Hathaway ***@***.***> wrote:
> […](#)

I'm also looking for this .bin file. Would love it if you could help me out.

jayhathaway, 2023-06-26:
https://drive.google.com/file/d/1fFEThG_DEakaaTcObgKXwwoApSy2G3RU/view?usp=drive_link

DocFreeman, 2023-07-17:
> https://drive.google.com/file/d/1fFEThG_DEakaaTcObgKXwwoApSy2G3RU/view?usp=drive_link
> […](#)

Much appreciated! I think the repo maintainers would appreciate it as well if you got rid of the link, airbreak does not distribute copyrighted files.

BravoIndia, 2024-06-14:
> I did flash my AirSense with my AirCurve's firmware, this worked perfectly and all the data structure definitions that the AirCurve had are present and operational. So let me be clear, this issue is entirely an artificial software limitation. If someone had an AirCurve 10 ST-A firmware this device would have all the data structure definitions and flashing its firmware onto virtually any other device, such as an AirSense 10 AutoSet, would enable a device that has 100% functional iVAPS and PAC modes. So if you have access to an AirCurve 10 ST-A, you can clone its firmware onto a base model AirSense or AirCurse device. In an assembly line situation, a machine can be disassembled and re-flashed with ST-A iVAPS / PAC ventilator modes in less than 10 minutes per machine per person.
> 
> The serial number and other vital product information are not stored within the firmware, that information is stored in an EEPROM; there is no uniquely identifiable information contained in the stock stm32.bin firmware file that OpenOCD dumps.

@nbritton Any chance I could contact you about your experimentation flashing AirCurve 10 S firmware onto the AirSense 10 AutoSet? Very much hoping you see this, thanks.