osresearch/airbreak #43: Brazilian Portuguese firmware

quilombodigital, 2020-05-04:
Hi,

I am from Brazil and I am trying to patch a brazilian portuguese firmware.
I don't know the firmware version because it does not show it (there is no about screen)
Like everyone I first stopped at the checksum issue, my checksum is ad5486f5c4721448893bc4edcb5905ba38b034ee54f00205bfcbe075fefab8e7.

I checked some offsets from the code and they seemed to match, so I decided to give it a try.
I changed the checksum and continued, to see how far could I go and I stopped at other issue:

_make: *** No rule to make target 'stm32.bin.xml', needed by 'stubs.S'. Stop._

Installed ghidra, analyzed using 'stm32 cortex little endian' , and generated XML and used it, but I am pretty sure this is not correct, since my guess is that the symbol names should come from somewhere else, and not auto generated, probably for offset matching. Then got stucked at:

_undefined reference to 'tim_read_tim5_

Trying to advance a little more, I commented the ghidrastubs part and flashed it.
Nothing looked different, no 'hacked' message, then I tried to enable 'all_menu' and some menu items in the menu changed, but no major feature enabled. 

Original Screens:
[image]
[image]
[image]

I need some guidance about what are the correct steps to adapt my firmware. I read the github issues but I am a little confused by everything.
Do I have to look for offset matches in version 401 and change the script? (I don't have a 401 firmware, I only have mine, what now?)
What about this ghidra stuff? what I must do inside it? where the names came from?
A small tutorial saying "1-match all offsets of blah to bleh, 2-convert xxx..." 

Thank you all for this incredible project.
The Brazilian health system is very precarious, you will be saving many lives.

dblunk88, 2020-05-04:
So, for the firmware, press the big round button and the bottom botton for a few seconds. This will take you to the clinical menu, where you can check your firmware version. Technically you do not need to compile to run it, just run the patcher and patch in the firmware... but there is a problem. You're running on neither 401 'nor 306 if that is your checksum... so you might have to find the offsets yourself.

tim5, from my understanding, calculates the current system time and if I recall correctly, there may even be the offset number (for 401) in the source... might be worth trying to rename the function in the XML (then again, I may be wrong).

gszakacs, 2020-05-04:
Finding it in the clinical menu is the easiest. I think you could also find it in the binary file at address 0x00040000.
[image]

dblunk88, 2020-05-04:
Oh nice. The patcher might work, change the checksum and give it a try. If you're compiling, give mine a shot:

https://github.com/dblunk88/airbreak/blob/master/patch-airsense

quilombodigital, 2020-05-04:
Hi,
The version is 306, so I used your script and it went normally. thank you!

Everything seems to have worked, and the modes are available, but when I choose ST or iVaps, It begins the screen, give a small blow, and then it restarts the everything and keeps restarting the machine. 

Any idea? Am I missing something?

[image]

gszakacs, 2020-05-04:
I have the 306 and the ivap mode works with a limited feature set on my 306. Some settings are not available such as backup rate.

Ideas: 
1. May I suggest that you remove the sd card ? Not sure if a corrupt sd card could trigger a restart but I did have an SD card failure the first time. 
2. I did notice the device resetting when I had the programming wires connected in the wrong order. Nbritton also had some issues (programming) when the comm wires were not connected properly.
3. The behavior may be related to the language setting. You may want to try  erasing all data, resetting the device, setting language to english.

quilombodigital, 2020-05-05:
I reset to factory but the problem persisted, then I removed the sd card and it **worked**. :) ST mode and iVaps. yeah!
Even without these features you cited, is it still usable as emergency/backup ventilator?
I know this is defined by a doctor, but just for curiosity, what would be the most used mode/settings for covid patients? I just would like to let it pre-configured by default.

nbritton, 2020-05-08:
Few things to think about.

1) I noticed on my machine that if I didn't have all the JTAG wires connected (especially the ground wire) the machine would sporadically reboot on its own.

2) Rebooting like that is often a sign of insufficient power, have you checked that the power plug is securely attached?

3) Are you powering your programmer using the CPAP machine? You're not suppose to connect the power pin when using a Raspberry Pi or other SBC / software bit-banger.

4) If you enable all menu items to always be displayed then you should be able to set the extra iVAPs settings.

5) Try erasing the SD card using the machine, the option is in the clinical menus near the very bottom.

quilombodigital, 2020-05-11:
The rebooting was pretty regular, when it started ivaps, one second later. My guess is that it happened when trying to register data at the SD card. It immediately stopped rebooting when I removed the SD card. Did not tried to format it, since I think I´will not need the sd card at all... :)

ruri-baka, 2020-05-11:
Is it only rebooting after the flow generator activates?