osresearch/airbreak #44: Binary disassembly and decompilation howto using Ghidra

nbritton, 2020-05-04:
The main chip on my AirSense 10 AutoSet is STMicroelectronics STM32F405ZGT6. This is an ARM Cortex-M4 32-bit ISA, however the Internet says the Cortex-M series only supports the 16-bit Thumb ISA.

STM32F405ZG Datasheet:
https://www.st.com/resource/en/datasheet/stm32f405zg.pdf

16-bit Thumb ISA Quick Reference:
http://infocenter.arm.com/help/topic/com.arm.doc.qrc0006e/QRC0006_UAL16.pdf

I was able to successfully disassemble and decompiled functions into C source code using Ghidra, below are the steps that I followed:

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

bdarcus, 2020-05-04:
This is great. But how does one apply this to enhancing the patch script; say adding missing menu items?

https://github.com/osresearch/airbreak/issues/42

dblunk88, 2020-05-04:
Thank you for the detailed guide, but I also think that you need to set the load address to 0x8000000

nbritton, 2020-05-04:
Yeah I think you could be right. 08008590 is a data block referenced by a pointer located at 08004128, and PTR_DAT_08004108 is referenced by FUN_08004100:08004100 and FUN_08004100:08004100. That appears to make more sense.

Are we able to share the decompiled source code? Just wondering on the best way to collaborate on deconstructing what the functions and parameter data are doing.

At the very least I'd like to figure out how to clear the gibberish that is displayed on my home screen.

gszakacs, 2020-05-04:
Not sure if I inderstand you correctly, but if if you are referring to gibberish as
[image]
then simply comment out the following two lines in the patch-airsense file:

FLAGSTR=$(printf 'FLAGS=0x%02x' $BUILD_FLAGS)
printf $FLAGSTR | patch 0x17588

COMMIT_HASH=$(git log -n1 --format=format:"%H" | head -c 7)
printf 'GIT=%s\x00' $COMMIT_HASH | patch 0x17764

wormyrocks, 2020-05-04:
Yes, those options are for debugging purposes - it gives an at-a-glance view of what version of the hacked firmware has been compiled as well as which UI settings have been enabled or disabled. Feel free to comment that out of the patch script.

nbritton, 2020-05-05:
Ok cool, I didn't realize those were intentional, I just thought it was some gibberish buffer overflow or something. I'm a principal computer systems engineer but this project isn't anywhere near my area of specialization so I don't have a lot of direct experience with reverse engineering.

jjsardar, 2023-12-14:
how change image in home ?