osresearch/airbreak #45: Accessing EEPROM data?

nbritton, 2020-05-05:
Hi, does anyone know how to access the EEPROM data? The product information that gets written out to the SD card within the file Identification.tgt appears to be coming from an EEPROM chip, "mmc:0:EEPROM" is what shows up in the decompiled source. I have my device in ASVAuto mode but ResScan v5.9 and OSCAR still show an AirSense device and one or both of those utilities appears to be getting confused about the device statistics because the stats are widely different.

I tried changing the following but it didn't have any effect about which device model was transcribed onto the SD card.
```
# Changes PCD value to 37216
printf '\x33\x37\x32\x31\x36\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' | patch 0x4020
# Changes product name to AirCurve 10 ASV
printf '\x41\x69\x72\x43\x75\x72\x76\x65\x20\x31\x30\x20\x41\x53\x56\x00' | patch 0x4030
printf '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' | patch 0x4040 
```

Here is the contents of my Identification.tgt file:
#IMF 0001
#VIR 0064
#RIR 0064
#PVR 0064
#PVD 001A
#CID <cellular device id? redacted>
#RID 000D
#VID 0027
#SRN <serial number redacted>
#SID SX567-0306
#PNA AirSense_10_AutoSet
#PCD 37028
#PCB (90)R370-7421(91)T2(21)85247141
#MID 0024
#FGT 24_M36_V39
#BID SX577-0200

I think it would be interesting to compile a list of output from the Identification.tgt file, in particular the PCB, PCD, and SID values. I belive PCB is the mainboard revision, PCD is the Device ID (i.g. 37216 is AirCurve 10 ASV machine), and SID is the firmware of course.

PCD Device ID Table:
USA: 37203, Canada: 37398 - AirSense 10 CPAP with HumidAir
USA: 37204, Canada: 37399 - AirSense 10 CPAP with HumidAir and ClimateLineAir
USA: 37205, Canada: 37400 - AirSense 10 Elite with HumidAir
USA: 37206, Canada: 37401 - AirSense 10 Elite with HumidAir and ClimateLineAir
USA: 37207, Canada: 37402 - AirSense 10 AutoSet with HumidAir
USA: 37208, Canada: 37403 - AirSense 10 AutoSet with HumidAir and ClimateLineAir
USA: 37209, Canada: 37404 - AirSense 10 AutoSet for Her with HumidAir
USA: 37210, Canada: 37405 - AirSense 10 AutoSet for Her with HumidAir and ClimateLineAir
USA: 37213, Canada: 37408 - AirCurve 10 S with HumidAir
USA: 37214, Canada: N/A - AirCurve 10 S with HumidAir and ClimateLineAir
USA: 37306, Canada: 37412 - AirCurve 10 ST with HumidAir
USA: 37307, Canada: N/A - AirCurve 10 ST with HumidAir and ClimateLineAir
USA: 37211, Canada: 37406 - AirCurve 10 VAuto with HumidAir
USA: 37212, Canada: N/A - AirCurve 10 VAuto with HumidAir and ClimateLineAir
USA: 37215, Canada: 37410 - AirCurve 10 ASV with HumidAir
USA: 37216, Canada: N/A - AirCurve 10 ASV with HumidAir and ClimateLineAir
USA: 28210, Canada: 28219 - AirCurve 10 ST-A with HumidAir
USA: 28211, Canada: N/A - AirCurve 10 ST-A with HumidAir and ClimateLineAir

nbritton, 2020-05-09:
I tried completely overwriting my AirSense firmware with my AirCurve's firmware, however, the device still retains the old serial number and AirSense_10_AutoSet header in the Identification.tgt file. So this information clearly must be stored somewhere else on the device.

ruri-baka, 2020-05-11:
Have you taken a look at this?

https://github.com/wormyrocks/airbreak/blob/master/tcl/airsense.cfg

Looks like you have to go into the debug console, but I bet its prolly sysconfig or optionbytes

nbritton, 2020-05-11:
I don't see anything significantly different within those dump files. For instance, optionsbytes.bin contains only a single address line that is different between the AirSense and AirCurve. Is it possible it could be stored on a different chip, such as within the STM8's flash?

ruri-baka, 2020-05-12:
> I don't see anything significantly different within those dump files. For instance, optionsbytes.bin contains only a single address line that is different between the AirSense and AirCurve. Is it possible it could be stored on a different chip, such as within the STM8's flash?

The STM8 is just the humidifier firmware I believe, but I haven't taken a look myself. Have you checked the srambkp and bootloader files? Does anyone know whats in bank1?

Asmageddon, 2021-03-01:
@nbritton @ruri-baka Did you happen to try to probe this any further? Any results?

nbritton, 2021-03-03:
Hi, sorry no, I lost motivation because I had some major life changes take priority.

acertain, 2021-10-02:
I've found airbreak unlocked VAuto firmware in ASVAuto mode mostly works but some stats are screwed up (e.g. pressure in overview & statistics tab), while ASV firmware works I think completely. All on an Autoset device. Haven't tried airbreak unlocked Autoset firmware.