osresearch/airbreak #47: Patching SX567-0302

grf100, 2020-05-09:
I tired using ruri-baka's 0302 patch script form [here](https://github.com/ruri-baka/airbreak/blob/SX567-0302-Support/patch-airsense). I get the HACKED! message on the screen, but it did not unlock any of the extra modes. The hash on the code I have is c7d16d912816b3e5b40158b695aaff629abc9f20bb4639943748bf2ae0d8b768
I assume since this hash is difference that the offsets are different? Any guidance how how to determine the offsets on the code I have?

ruri-baka, 2020-05-11:
What model and PCB revision hardware are you testing on?

grf100, 2020-05-11:
Aircurve 10 Vauto, model 37051, PCB is Rev1.0

beast-in-black, 2021-05-13:
Sorry to necro this almost exactly a year later... I found that the 0302 patch script itself needed to be patched to get a successful flash with the new menu items (just flashed it successfully but haven't tested the new modes yet).

Attached is my modified patch based on @ruri-baka's original patch.  

[sx567-0302.patch.txt](https://github.com/osresearch/airbreak/files/6475445/sx567-0302.patch.txt)

Apply using `patch -p1 < sx567-0302.patch.txt`

TexNOz, 2021-12-15:
@beast-in-black   How did this turn out for you?

beast-in-black, 2021-12-15:
@TexNOz I've been running the patched firmware right from when I posted the
patch. Works great, including the new modes. I've kept mine on APAP mode
just to be conservative although I have increased the max pressure limit to
23 (stock tops out at 20).

ArthurMafra, 2022-03-09:
> What model and PCB revision hardware are you testing on?

Hi @ruri-baka , Let me ask you a question, how do you get to generate corresponding stubs.S for your device. I do have firmware 306 but using 401 stubs.S caused problems. Have you managed to run your airsense with sd card inserted? Vsauto, S, ST are they working? With SD card inserted I get device on infinite restart loop. SD card cannot be left on with machine running. Remove sd card allow it start but modes are not working correctly. I believe it is due to incorrect stubs.S file. What is your input? I know long time passed but I could not find 401 firmware anywhere nor generate stubs.S for 306.
Regards

ArthurMafra, 2022-03-13:
> Aircurve 10 Vauto, model 37051, PCB is Rev1.0

Hi @grf100 I know the project has been abandoned, specially with new releases getting out Airsense 11 and future aircurve.
I have been donated Airsense to treat my apnea from person here from github. Is it possible for you to share your aircurve 10 firmware since it does not contain any identifiable information? It has been noted by nbritton and also confirmed by the fact that checksums for each version always match. Thanks for much either way. Contact arthur iphone @ gmail dot com.