osresearch/airbreak #50: Help : tcl/airsense.cfg:75: Error: invalid command name "mrh"

DiegoGiovany, 2020-07-22:
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.101916
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
tcl/airsense.cfg:75: Error: invalid command name "mrh"
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "tcl/airsense.cfg", line 75

DiegoGiovany, 2020-07-22:
any ideas?

DiegoGiovany, 2020-07-27:
SOLVED: use updated version of openocd, not the one installed via Mac brew

Grawer, 2021-01-10:
Just a note, Open On-Chip Debugger 0.10.0 (2020-12-28) works fine, while base 0.10.0 version doesn't.

beast-in-black, 2021-05-13:
openocd v0.11.0 (built from the source of the `v0.11.0` tag) also works fine.