osresearch/airbreak #51: Possible improvement to PAP firmware itself.

Asmageddon, 2021-05-14:
For context, I am a Sleep Disordered Breathing patient, and active in a community where we discuss treatment and surrounding research.

It's a major grievance to many of us that whereas PAP machines are generally sufficient treatment in older men, younger people, women, and milder cases(where no degeneration of arousal mechanisms is present) experience only very limited benefit - of which some is inherent to the limitations of NIV, while some is regrettably down to algorithmic matters, with ResMed doing much worse than its competitor, Phillips Respironics.

Specifically, the following features would likely make a large difference:

 * Exhale Support - something similar to [Phillips Respironics FLEX features](https://cdn.discordapp.com/attachments/749382095113289800/829255676764094504/unknown.png), which essentially reduce pressure during exhale to alleviate expiratory overpressure buildup contributing to exhalatory intolerance without compromising the capacity of treatment to stent airways open. This would especially be important for a large share of patients who have narrow nasal cavities.
 * Pressure support waveform contour modification - while ResMed's EasyBreathe is very tolerable, it is not as effective as steeper pressure support curves, both in terms of preventing the need for the body to escalate respiratory effort, and in terms of elevating airway pressure during inhalation to maintain the capacity of treatment to stent the airway.
 * Disabling ASV mode backup rate - persistent, neurological central apnea is rare in SDB patients, whereas treatment-emergent central apneas diminish and disappear as the body adjusts to treatment, with the backup rate instead picking up post-arousal drops in ventilation caused by a drop in CO2, and preventing transition back to voluntary breathing control.
 * Making it possible to use a narrower min and max PS range in ASV mode, rather than mandating max PS to be at least min PS plus 5cmH2O

Of course, this is not a finite list, and much more could be improved, but past the above, I'm not sure if there's anything I'd guess likely to be possible to easily integrate into existing firmware via reverse engineering.

Drmaestro, 2021-05-14:
Hi,

You might have observed that there has been no recent activity by the developers for a long time and as such, it seems that the project is practically abandoned. I had contacted one developer 8-9 months ago and he told me that there was no real reason to go on with the project as the main aim was to prove that a CPAP unit can be used as a ventilator in an emergency situation (and they did prove that by the modification) and also the projected ventilator shortages never happened. Altering the main firmware for other purposes, like enabling other modes, would probably create legal troubles for them (copyright issues and also potential harm to patients using these mods). Of course, new developers can always continue from where they left, but from what I have observed in the last 9 months, there is very little to almost no interest in the project. (I don't want to create an air of negativity and would be very happy to see some more activity but that seems to be the truth for now).

The exhale support already exists in Res Med 10 devices, however it is limited to 3 choices (1-3 cm H20). As such, it is a limited form of BiPAP. If you want more flexibility, then the manufacturer wants you to buy their BiPAP device...

Asmageddon, 2021-05-14:
Hi @Drmaestro, thanks for the reply. The airbreak project already does unlock all modes, a feature I've taken advantage of. Unfortunately, the Expiratory Pressure Relief feature is not actual exhale support, but rather a limited form of pressure support(which is essentially inhale support). I already run my device in bilevel mode, but... it's not really enough for my case, nor for the cases of many other people. The worse nasal breathing, the more EPAP is needed, and the less EPAP can be tolerated. I had high hopes for ASV, but the backup rate just hurts the efficacy so badly. 

Ultimately, PAP is merely management, and the ultimate goal is skeletal surgery(and/or whatever adjunct surgeries are necessary), but not everyone has access to such treatment. Improving the algorithms on ResMed devices could potentially save the lives of at least a few of the people I know from inescapable fatigue.

zhuhongchao, 2021-06-27:
> Hi @Drmaestro, thanks for the reply. The airbreak project already does unlock all modes, a feature I've taken advantage of. Unfortunately, the Expiratory Pressure Relief feature is not actual exhale [...]

I have followed the instructions to unlock the ASV and other modes, but the parameters in the menu are incomplete and some of them cannot be set, did you solve this problem? (trigger, cycle, Backup rate, etc)

Asmageddon, 2021-06-30:
The ASV mode doesn't allow configuring trigger and cycle, and I believe the backup rate setting is for iVAPS, not ASV, in which backup rate is mandatory. I've personally not had success with the unlocked ASV mode.

jlm255, 2021-09-20:
Asmageddon, I am an SDB sufferer as well, RERAS, not apneas, and I was hoping to use this firmware to be able to fiddle with the EPAP beyond the EPR of 3 for the Airsense 10. Would it accomplish this?

Thanks!

chuckhacker, 2021-11-14:
Trigger and cycle are for BiPap modes:

https://www.youtube.com/watch?v=ts9lNJ2g1IE

Asmageddon, 2022-04-23:
I've recently come back around to this idea, particularly hoping to disable the ASV mode backup rate, which I hope could help me address my horrible residual sleep fragmentation on BiPAP modes.

@osresearch @wormyrocks @colinoflynn If you could help me with any of the notes not uploaded to this project's github, or pitch some ideas about how I could possibly approach debugging a running device to identify the code path or variable responsible for this, I would be deeply grateful.

macka-the, 2024-02-05:
Hi, I am sufferer of same sleep fragmentation and been wringing my machines out and closely looking at algorithms. As well as an AS10 I have an interesting machine I have not seen many investigating yet. It is a Lowenstein Prisma 20a. I believe you may be starting with an inferior base for all your efforts. The Prisma is equipped with way better hardware and design IMO. The algorithms are way better as well. The company has built its knowledge and I believe it algorithms on hospital ventilators and it shows. A base model APAP/CPAP offers 3 comfort setting which leverage bilevel and trilevel modes of higher models. The most interesting is the trilevel which they call softpap 3. This mode gives pressure support of 4 and switches up the algorithm as different stages are reached in pressure. Also the algorithm delivers ipap pressure by the time inspiration begins not mid-way or even at the end like Resmed and others. This makes a difference in smoother delivery at more appropriate times which benefits narrow nasal cavities rather than the late rush of air which just aggravates as pressures rise.
The company has a flashing tool and all models in a range have one firmware. You can see the vacant menu locations. Parts of higher modes are used for comfort provision.

jlm255, 2024-02-05:
Sounds intriguing, where/how can I get one? I just buy my machines, going
through insurance is a nightmare.

macka-the, 2024-02-06:
I am in Australia, we have to buy our machines. Problem is some of these companies come from hospital ventilator and associated equipment industry IMO. There is a lot of price gouging going on, maybe because of the Insurance model in other countries as well. Government health is huge business. Look up the ventilator contracts Phillips had with US gov. Money thrown at them and then the abuse is incredible. 
Anyway, you can get the German made Prisma in Australia and Europe, but the cheapest by far is if you can get one out of India. I would like to get my hands on the PrismaLab. It is the top model with every mode. It has real time data monitoring and ability to control via network with smoothed transitions so a sleep tech can move you between modes without disturbing you.
These are unavailable in the US. They are far superior IMO. They are being kept out by FDA I believe, must have not greased the right palms.

jlm255, 2024-02-06:
Dang that sucks. I'd like to get your contact info somehow.

I bought a Resvent from China because I wanted to try volume assured (iVaps
EVaps whatever). It basically keeps my pressures near 4 but somehow forces
a breath if my breathing drops below a set value. This seems to be just as
effective or more so, but what I think I am finding is that my deep sleep
requires different values than REM so a better algorith or sleep tech to
figure things out sounds great.

I have had UPPP surgery, MMA surgery, and 2 rounds of tongue radiofrequency
ablation and I think my apneas are cured but without a doubt I still have
some strange breathing problem. After all the surgeries, I tried going
without CPAP for a month and eventually things were going downhill again.

My sleep and life have improved a lot. Before all this my life was really
really in a dark spot. Everything was falling apart (job / family / etc)

jlm255, 2024-02-06:
I think for the first half the night I'm better off without the CPAP. But
then the REM half I need it desperately.

Asmageddon, 2024-02-17:
> The company has a flashing tool and all models in a range have one firmware. You can see the vacant menu locations. Parts of higher modes are used for comfort provision.

Wait, are you saying the prisma devices are hackable by default? 

Anyway, I've experimented with both trilevel and with fasted PS rise time in my custom firmware, the problem with fast rise time is that it becomes almost impossible to detect the actual end of breath, so you end up causing horrid asynchrony, and I wasn't able to solve this problem. Trilevel actually improves comfort, but also has some challenges.

I'm rather interested in Prisma's rapid pressure response due to a valve-based design, but if they solve the two above issues, their devices might actually be great.

macka-the, 2024-02-17:
> Wait, are you saying the prisma devices are hackable by default?

Maybe, all I know is anyone can see the link to firmware updates for every model and a link to a flashing tool of some type. There is an unused micro usb port on the machine. Communications modules are available.
I have a lower end model and it has bi-level and trilevel function masquerading as comfort features. The menus have lots of spare spaces and all models in that one range share user & service manuals.
Don't know what protections are at chip level but the service manual says to enter service mode and simply enter serial number on side of machine when performing a mainboard replacement

jlm255, 2024-02-17:
How fast is fast? I'm currently using 900ms

jlm255, 2024-02-17:
My breathing slows down too much during sleep so I have back up breathing
rate, fiddling with rise time, and the length of time CPAP machine waits to
increase pressure to try to force a breath (on resvent they label it Ti
which is confusing) I was wondering if instead of doing this, what if we
breathed more concentrated oxygen. Would this make up for the slow
breathing rate? Thoughts please.

jlm255, 2024-02-18:
> Unless you have neurological sleep apnea, do not enable backup rate. If
> your body pauses breathing, it's doing it because you are OVER-ventilated.
> Breathing is supposed to be slower and shallower during sleep.
>
> Also, lack of oxygen is not the problem. The problem with sleep disordered
> breathing is the sleep fragmentation caused by the body's reaction to
> escalating airway resistance. Oxygen doesn't matter much.

I would have said the same thing, but recently I was reading this guys blog
and it rocked my world:

https://www.cpaphealthissues.com/2019/06/ideal-body-weight-cpap-bilevel.html

I am someone who has never registered an AHI over 2 (on the CPAP) from
before my sleep apnea diagnosis, and through surgeries, and to the present
day. My sleep lab test showed an AHI of 7 and an RDI of 30. So in general
my obstructions have always been pretty minimal, but it has been
fragmenting the hell out of my sleep since birth. If you look at my fitbit
data, which I know isn't perfect, you can see how fragmented my sleep is.

My pressures are usually from 4 to 6, I actually think that a pressure of 2
might be enough for me, but very hard to find a CPAP that goes that low.
There is no question in my mind that when the CPAP is blowing air into me
like crazy that it also fragments my sleep.

While I am not positive yet, fiddling with Ti and breathing rates etc does
seem promising. I have had some phenomenal days where I didn't know it was
possible to feel so happy. But I can't get it to 7 days a week. I would say
on average I can get one "perfect" day a week, 5 very functional days, and
1 day where I am very "off" and sometimes "barely surviving" but it seems
like those days are getting less frequent as I seem to be honing in on
which numbers will really prevent a disastrous night.

macka-the, 2024-06-23:
@Asmageddon 

I now have Prisma30ST I found cheap. It has a mode aPCV I was after for same reasons as you. What I didn't know I was getting as well, was all the usual S T ST & auto ST modes, but with the addition of volume assurance on each of them.

So you can set your usual bi-level mode but with a Vt target to be maintained. The sensitivity is driven by looking at every single breath, 5 breaths, or 8 breaths. Timing on trigger, InRamp, exRamp. Choice of 3 EEPAP levels or none. Can have a variable ps and epap.
Backup is part of aPCV and I have worked around it set at 10. It has come in handy for carrying me through a flow limited period and can save the arousal.

I have the 25ST as well and it offers everything without the volume assurance. Still very good in auto S/T mode. I have found the backup quite handy in this machine set to 10 also. Synchronicity is nowhere near as easily broken in the Lowensteins as Resmed imo.
The machines seem to follow and adapt very well to erratic sponteanous breathing and then pickup again. The timing adjustments are also very good. EEPAP is essential imo to solving a lot of issues. I also use a series of flowrate hacks to get that at tolerable levels then adjust machine from there. I have intolerance for flow over a certain rate. If you breach this rate there is nothing can be done to make therapy workable. I believe have seen this in many individuals. You must approach flowrate as any other adjustment that has a perfect setting for every individual.

My next project is to get a hold of a Luisa. Ultimate in home ventilators. 

https://medbidding.com/shop/secondhand-mechanical-ventilator/secondhand-home-mechanical-ventilator/secondhand-lowenstein-luisa-lm150td-home-mechanical-ventilator/

Asmageddon, 2024-07-12:
@macka-the Oh wow that sounds amazing. Btw I accepted your discord friend req.

aske-cph, 2025-06-19:
Thank you @macka-the for the very valuable info!

Unfortunately the 30ST is very difficult to find and the LAB i've not even seen for sale, it's weirdly either 25S or the Vent-30 i can find at reasonable prices in EU through German e-bay mostly. 

Is the 25S missing crucial features like the EEPAP, it's difficult to gauge the exact specifications from Lowenstein's own website?

I've only found a few 30ST's on a Turkish site called Medbidding that looks a bit sketchy and has no Trustpilot. 

Thank you so much in advance!

aske-cph, 2025-06-19:
@macka-the Thank you!  30ST it is then if i can find it below 2000 euro. Too bad the 25S is not up to par as it is a lot cheaper and more available - and incredible a few simple features cost several thousand euros, we really need to be able to flash these machines, especially if a Smart Max could more or less be made into a lab with different firmware.

realdannys, 2025-06-26:
@macka-the - you got me excited further up the page making think I could flash my Prisma20A!

I need ASV unfortunately - but I purchased the Prisma20a before I knew this. It's such a shame it's totally capable of ASV but it won't run it. An absolute shame.

macka-the, 2025-06-27:
@realdannys Are you sure you need ASV? I have helped a lot of people off ASV who only had induced TECSA with bad settings on a regular machine. The type is usually those with a low tidal volume due to anatomical restriction that has developed or induced by chin tuck, back sleeping and tongue. The constantly low tidal volume, while sleeping, makes you prone to a protective reduction in respiratory drive after the larger post arousal breaths that come after spells of insufficient tidal volume. The sensitivity gets worse over time if the low tidal volume is not dealt with. The popular belief that lowering pressure and removing or lowering pressure support (like epr) exasperates the problem. You need these things to raise tidal volume properly.
Clinically, setting a machine to APAP mode unduces this in these types, the response then to declare you mixed apnea and throw you onto an ASV, set that with wide settings that will cover up the centrals that it will ultimately cause. Not a fun journey.
If you have neurolgical central apnea then that is entirely different.

DrSkittles, 2025-07-28:
@Asmageddon  I understand the Airbreak Plus project is unfinished and untested, but would it be foolish of me to try to use it at this stage?

I tried out the VAuto mode last night, and while this could definitely be placebo, it seemed to be much more comfortable using your modified firmware compared with the original VAuto firmware. I was able to bump my inspiratory pressure up to 11.6 without too many leaks or aerophagia, while with the original firmware and same exact settings, these problems are significantly worse and seriously affect my therapy. Obviously I can't set TiMin/Max or cycle/trigger on the Airbreak Plus firmware, but every other setting was the same.

I've been through roughly six machines/firmwares now, and I seemed to have the most success using yours as I'm trapped between a rock and a hard place regarding my therapy; need higher inspiratory pressure, but just before inspiratory pressure reaches a therapeutic point, aerophagia and leaks wreck me, and most external measures to reduce those two problems have failed. Need higher inspiratory pressure to achieve a higher tidal volume, and need higher inspiratory pressure to prevent flat/double peaks related to UARS, but luckily hypopneas are completely eliminated around IPAP of 8.0 so they aren't an issue.

Not looking for PAP advice or updates to Airbreak Plus, especially since I know you were tuning it for your specific needs, just wanting to know a bit more about where the project currently is so that I know if what I'm experiencing is placebo/luck.

JohnnyDoe777, 2025-08-03:
Hello @Asmageddon,

I tried to build the new firmware from the latest commit 5d64c73, but it gives me the following error:
`patches/common_code.h:239:10: fatal error: feat_triggercycle.h: No such file or directory
239 | #include "feat_triggercycle.h"
         ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`
When building from commit dafa5cf from Oct 4, 2024, the build was successful and worked well. I have tested VAuto and ASV modes. Both drew nice graphs which are not present in the stock firmware.

I read that you won't be present for two weeks, so enjoy your vacation or whatever you have planned. 
Your refined VAuto also seems gentler to me and allows for a slightly higher pressure without being uncomfortable. I also like the idea with EEPAP. I will try it for a few nights on VAuto and ASV. Then I would like to try your new version of ASV that you are planning to commit. Thanks

aske-cph, 2025-08-03:
@macka-the regarding your initial comment about tweaking Lowenstein instead of Resmed - i took your advice and bought myself a Prisma CR - and it's on the way - hope it works. 

Do you have experience coding? 

I've already looked at OSCAR and understood some of the loaders / data schematics and look forward to getting it to work with OSCAR first and foremost, but also understanding the difference in firmware/hardware - as you described the CR has all the modes of 30ST, 25S and down right? Just with an addition of an ASV mode?

Maybe a seperate project just for Lowenstein devices should be created to tinker with those and understand them in depth. All within legal bounds of course!

aske-cph, 2025-08-03:
If that is the case  i hope you can tweak the settings to mimic bilevel if needed - not sure exactly what i need to eliminate FL with OSA/UARS mix. But lets see when i get it in a week.

Yes it was very cheap because it's used a lot i think (sub 500 euro ), not exactly sure, they say it works and the seller is very reputable. 

But i have already fixed a used 20A i bought for 100 euros before that was never used but gave error codes - just opened the case, read about various error codes, and found it it was related to the blower, so opened further to the Micronel blower and got it working by reattaching the cables. 

So i feel a bit more confident about the project and will acquire further models if needed.

Asmageddon, 2025-08-13:
@JohnnyDoe777 Apologies, I should have double-checked my commit. I have uploaded the missing files now. I recommend using the lowest cycle sensitivity setting in S mode if you're struggling with premature cycling(short inhales followed by exhales that are flattened or have an upwards spike in the middle instead of the normal shape)

Asmageddon, 2025-08-28:
@JohnnyDoe777 I haven't compiled it in a while, but the repository is up-to-date with my local branch, which is what I have on my own device. It sounds fine. Also, I believe the VAuto+ mode has an ASV mode, if you set TiMin to minimum. It should also work better than the normal ASV.

@macka-the You can alter essentially everything you want. The (fairly simple) `squarewave.c` file contains a full replacement for a control loop, and you can implement anything you want instead. 

Alternatively, in the `wrapper_limit_max_pdiff.c` file, you could write a replacement for any of the modes - it overrides a function that limits how much the pressure can change from one tick to the next, and seems to be one of the last functions called before the commanded pressure value is sent to the blower. E.g. any `*cmd_ps` value you write there is what the pressure will be set to.
Using that, you are free to read any of the memory values I've mapped out in `common_code.h`, which includes settings for most of the modes - pressure, flow, baseline- and total- leak-compensated flow, leak, bla bla.

TLDR: Yeah you can write any custom algorithm you want to replace(or overlay on top of) any of the modes (maybe except the ASV ones, they seem to operate on different code paths)

AJolly, 2025-12-16:
@Asmageddon consider enabling issues on your repository so discussion can be held there?  And suggested machines for someone just starting out on their sleep improvement journey, and where to best source them? (I used to be a reverse engineer a long time ago, so I'm comfortable hacking away)

AJolly, 2026-01-18:
@Asmageddon thanks! im going to acquire an 10 or two.  my official sleep study came back negative, but im still seeing crazy hr spikes and disordered breathing (measuring via an polar h10 / emay breathing monitor), so I'm suspecting UARS