osresearch/airbreak #52: line 63 postboot error invalid command name (Help me)

msdgd, 2021-09-07:
I followed all the instructions, such an error came in the flashing part. device looped in "resmed" splash screen. Installing the backup 32.bin fixes it, but I couldn't hack it, can you help me?  @nbritton @enjrolas @redpath  @osresearch  @colinoflynn @graver  @beats-in-black 

https://gfycat.com/warpedreflectingbarasinga

`[root@fedora scripts]# telnet localhost 4444
Trying ::1...
telnet: connect to address ::1: Connection refused
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> flash_new stm32-unlocked.bin
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Tickling watchdog...
device id = 0x10076413
flash size = 1024 kbytes
Resetting device...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Overwriting flash memory...
device id = 0x10076413
flash size = 1024 kbytes
Verifying that flash was written properly...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Resetting watchdog bit...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
Writing magic numbers back into RCC memory...
Booting device...
tcl/airsense.cfg:63: Error: invalid command name "postboot"
in procedure 'flash_new' 
at file "tcl/airsense.cfg", line 63
`

msdgd, 2021-09-07:
If I send you my st32.bin file via e-mail, would you mind if you patch it and send it?

msdgd, 2021-09-07:
It gives the same error when restoring/flashing the stm32.bin file, but the device returns to normal. restarts are interrupted. 

`flash_new stm32.bin
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Tickling watchdog...
device id = 0x10076413
flash size = 1024 kbytes
Resetting device...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Overwriting flash memory...
device id = 0x10076413
flash size = 1024 kbytes
Verifying that flash was written properly...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800021c msp: 0x2000f000
Resetting watchdog bit...
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
Writing magic numbers back into RCC memory...
Booting device...
tcl/airsense.cfg:63: Error: invalid command name "postboot"
in procedure 'flash_new' 
at file "tcl/airsense.cfg", line 63
> 

`

msdgd, 2021-09-07:
It is normal?

`─(kali㉿kali)-[/media/kali/Lexar]
└─$ sha256sum stm32.bin
533b91127aa22e05b933db203ad56c449dc12a8c3fd62f57bd88c472a8061775  stm32.bin
                                                                             
┌──(kali㉿kali)-[/media/kali/Lexar]
└─$ sha256sum stm32-unlocked.bin
3ef08b46768c9a3bb46805c61c41464023f7b18f8deee102946bf4c7ceaa1255  stm32-unlocked.bin
`

msdgd, 2021-09-07:
or Can you send me the ready stm32-unlocked.bin file to I make flash

msdgd, 2021-09-08:
or there is no problem in the patching part, is there a problem in the flashing part?  because it gives postboot error 

`[root@fedora scripts]# ./patch-airsense stm32.bin stm32-unlocked.bin
stm32.bin: Tamam
patching 000000f0
patching 00017500
patching 0001a540
patching 0001b860
Including breath.bin
patching 000bb734
Including graph.bin
patching 000fd000
patching 000f9c88
patching 00004fa8
patching 00004fc4
patching 00007eb0
patching 00007ee8
patching 00007ecc
patching 000084a8
patching 00008590
patching 00066470
patching 000050d0
patching 0000817c
patching 00008198
patching 00004f9c
patching 00004fb8
patching 00007ec0
patching 00007f14
patching 00007ef8
patching 00007f30
patching 00004fd4
patching 00007f68
patching 0000802c
patching 00008048
patching 00008064
patching 000080b8
patching 0000809c
patching 000080d4
patching 000080f0
patching 00008128
patching 00008144
patching 00008160
patching 00017588
fatal: bir git deposu (veya üst dizinlerinden birisi) değil: .git
patching 00017764
3ef08b46768c9a3bb46805c61c41464023f7b18f8deee102946bf4c7ceaa1255  stm32-unlocked.bin
`

jasonminsookim, 2021-09-24:
Hey I have had the same exact problem as you. I've used the pull request from dblunk88 with SX567-0306. I'll keep trying to debug this.

jasonminsookim, 2021-09-24:
@msdgd I think I finally solved it! Here are the steps that I took to get it working.

1) Instead of cloning this git repo, clone the pull request from @dblunk88
```
git clone https://github.com/dblunk88/airbreak
```

2) I also had to comment out the path_code function in the path-airsense script
```
# patch_code() {
# 	#if you want to add the extra breath mode
# 	BUILD_FLAGS=$(( BUILD_FLAGS | (1 << 0) ))
# 	if [ -r breath.bin ] && [ $version = "401" ] ; then
# 		echo "Including breath.bin"
# 		cat breath.bin | patch 0xbb734 || die "binary patch failed"
# 	elif [ -r breath.bin ] && [ $version = "306" ] ; then
#                 echo "Including breath.bin"
#                 cat breath.bin | patch 0xbb4d4 || die "binary patch failed"
# 	else
# 		die "breath.bin not found, did you run make?"
# 	fi 

# 	if [ -r graph.bin ]; then
# 		echo "Including graph.bin"
# 		cat graph.bin | patch 0xfd000 || die "graph patch failed"
# 		printf '\x01\xd0\x0f\x08' | patch 0xf9c88 || die "graph fixup failed"
# 	fi
# }
```

3) Follow the instructions on https://github.com/osresearch/airbreak/blob/master/docs/firmware.md

ArthurMafra, 2022-03-09:
> It is normal?
>
> `─(kali㉿kali)-[/media/kali/Lexar] └─$ sha256sum stm32.bin 533b91127aa22e05b933db203ad56c449dc12a8c3fd62f57bd88c472a8061775 stm32.bin
>
> ┌──(kali㉿kali)-[/media/kali/Lexar] └─$ sha256sum [...]

Hi @msdgd , do you have 401 firmware? Can you share over mail of google drive? I would like to have the OEM 533.. checksum.

msdgd, 2022-03-11:
I would love to share but they said it would be a problem due to the serial number and differentiation of the device

closed 2022-03-11 by msdgd

ArthurMafra, 2022-03-11:
> I would love to share but they said it would be a problem due to the serial number and differentiation of the device

There is no problem, serial number information is not stored inside firmware as nbritton noted. I will mainly use it to fix my device. I can give you a email so you can send me the download link. Is it possible for you?

ArthurMafra, 2022-03-11:
@msdgd  Note @nbritton comment.  Please share at arthuriphone@gmail.com
[image]

adrianbureu, 2024-04-07:
I’m having this same issue. Followed the instructions by @jasonminsookim but im still getting the same error. Any ideas? @msdgd

msdgd, 2024-04-07:
> I’m having this same issue. Followed the instructions by @jasonminsookim but im still getting the same error. Any ideas? @msdgd

So, it's been a long time and I don't know the exact details. But I tried it on several virtual machines with windows. Fedora, Ubuntu, Kali Linux... Which operating system are you trying on? You also made this instruction "git clone https://github.com/dblunk88/airbreak", right? 
I remember I was trying to do some things from scratch with a clean installation.

adrianbureu, 2024-04-07:
> So, it's been a long time and I don't know the exact details. But I tried it on several virtual machines with windows. Fedora, Ubuntu, Kali Linux... Which operating system are you trying on? You [...]

Yes, I git cloned directly from there. I’m using a fresh installation of debian on my raspberry pi. But I’m skipping the patching script, just trying to flash a firmware I have.

msdgd, 2024-04-07:
how? Instead of patching your autoset stm32 file and flashing stm32.unlocked, are you flashing the vauto/asv/st stm32 file of another device instead of autoset stm32?

adrianbureu, 2024-04-07:
> how? Instead of patching your autoset stm32 file and flashing stm32.unlocked, are you flashing the vauto/asv/st stm32 file of another device instead of autoset stm32?

yes exactly, I have an asv file that I am trying to flash.

msdgd, 2024-04-08:
> > how? Instead of patching your autoset stm32 file and flashing stm32.unlocked, are you flashing the vauto/asv/st stm32 file of another device instead of autoset stm32?
> 
> yes exactly, I have an asv file that I am trying to flash.

I'm also curious about the procedure you requested, but I didn't do it. I patched my own file normally. I think flashing the file of another machine definitely requires more than the steps in this regard, since the serial numbers of the device etc. are different. So I dont know