osresearch/airbreak #54: Bricked SX567-401 with unusual hash

chuckhacker, 2021-11-14:
I used Windows tools to dump the firmware. My SHA256 hash was: aa79d2e4f9114f8f9162b52a321eb4cf4d123db9ad55d4c2e08b5716fcaea25f

This is a different hash than the SX567-401 expected in the patcher. I saw that the offset 0x8590 had the expected 0x3 value in my dump, so I modified it and reflashed, expecting to get all of the modes. A one-byte change with expected value should be harmless, right (famous last words).

When I reflashed using the Windows tools, the LCD screen started blinking all white and off with the backlight upon power on. Then when I then reflashed the original, the only thing that happens is the green power LED on top flashes -- it is otherwise dead. I looked to see if others have the same issue before with original stock firmware, and the only thing I could find was someone who had a partially connected humidifier. I tried powering on with the humidifier power disconnected and also reconnected, to no avail.

I'm concerned that the erase before write might have also taken out other parts of the memory? Otherwise, I'm not sure why it is now bricked...

chuckhacker, 2021-11-14:
Is there any JTAG / hardware debugger worth taking a look at?

cr2551, 2021-12-07:
I also used windows to dump the firmware, i got the same version and the same hash as you.
I only had trouble with the part of the code that checks to see if the hash is the same. I just commented out those lines. However, i do not get the extra options like easy breathe or back up rate. I only get to adjust the pressure settings except for Ivaps mode where i have settings for target Va and pt.

darkflame808, 2022-08-28:
Oh boy. It's been quite some time since this thread popped up. However I would like to resurrect and say that I am in the same boat as you. Except I am not using Windows. I am using openocd on a M1 Mac Pro 16". My hash matches yours exactly. Could this be that rested updated firmwares without updating the versioning? To throw a wrench into this project?

Just curious if. you made progress. I have two more machines I am going to dump to see if they match. My gf gave me the thumbs up and I also have a spare unit I bought in case this one kicked the bucket. That one has never been powered up yet. I'll report back.

Also, if it is of any help, my gf has a aircurve isv unit. I'll dump that too! lol.