osresearch/airbreak #61: AirCurve 10 ASV 401 firmware

Millstone50, 2023-08-31:
Does anyone have this stock firmware in their possession? If so can you leave an email or something where I can get in touch?

mtemple, 2023-12-02:
Interested as well!

dmlerner, 2023-12-22:
Unmodified SX567-0401 firmware read off an actual ASV: 

https://drive.google.com/file/d/1AkftFeG2ySBn1yH07WUegWm5Q3COc7oA/view?usp=sharing

kqj2354olkjawef984, 2023-12-26:
Looks like the comment was deleted - here's a (new) link: https://file.io/EPyfOoyMuFlG

EdWorth120, 2024-01-29:
Hello @Millstone50 @kqj2354olkjawef984 and @mtemple, does anybody still have that firmware, the present link doesn't work anymore.

shaareable, 2024-01-30:
I have an ASV however I do not know how to extract the firmware for you.

kqj2354olkjawef984, 2024-01-30:
ASV 401 firmware: https://fastupload.io/SQSUzsOJ0P9zJyY/file

edmundo, 2024-01-30:
> ASV 401 firmware: https://fastupload.io/SQSUzsOJ0P9zJyY/file

Thank you

EdWorth120, 2024-01-30:
> ASV 401 firmware: https://fastupload.io/SQSUzsOJ0P9zJyY/file

Thank you

closed 2024-02-17 by colinoflynn

colinoflynn, 2024-02-17:
There are no rights to redistribute vendor firmware, please do not ask or use this repository to do so if you want the project to continue to be hosted here.