osresearch/airbreak #62: Aircurve 10 Vauto US firmware

zaienk, 2024-01-03:
Hello,
I'm looking for the firmware of the Aircurve 10 Vauto US firmware which provides access to the trigger parameter.
If somebody have it and can share it with me.
Thank you

EdWorth120, 2024-01-29:
Hello @zaienk here is a link: https://file.io/iUANwgVq0Pb1

zaienk, 2024-01-29:
Thanks a lot !!!

kqj2354olkjawef984, 2024-01-30:
I see the link has been deleted. Care to repost? Here is vauto firmware, without the trigger parameter. https://fastupload.io/yp1SXPNKCu3bYo5/file

edmundo, 2024-01-30:
> I see the link has been deleted. Care to repost? Here is vauto firmware, without the trigger parameter. https://fastupload.io/yp1SXPNKCu3bYo5/file

This is the link to Vauto version 401 with the trigger parameter: https://fastupload.io/en/To6uZ2TGrwHqjJp/file

EdWorth120, 2024-01-30:
> I see the link has been deleted. Care to repost? Here is vauto firmware, without the trigger parameter. https://fastupload.io/yp1SXPNKCu3bYo5/file

This is the firmware of Vauto with the trigger parameter enabled: https://fastupload.io/en/To6uZ2TGrwHqjJp/file

colinoflynn, 2024-02-17:
There are no rights to redistribute vendor firmware, please do not ask or use this repository to do so if you want the project to continue to be hosted here.

closed 2024-02-17 by colinoflynn