osresearch/airbreak #65: Patching firmware of existing ASV

cpapenjoyer, 2024-02-24:
Hi, I have an ASV and I'm trying to jailbreak it so I can set the maxPS value to 5 while keeping the minPS at 3.
Would I be able to achieve this by following the standard firmware patching instructions?
(By patching the official firmware of my ASV device)
Will it still function the same way, but with more freedom to set maxPS?

For more context, the ASV has solved my CA events but I've noticed my heart rate goes up whenever the PS ramps up
Thanks :)

smhc, 2024-04-09:
Were you ever successful in doing this?
I attempted to patch the 305 ASV firmware with the patches from:

https://github.com/Asmageddon/airbreak-plus/blob/master/patch-airsense#L149
https://github.com/Asmageddon/airbreak-plus/blob/master/patch-airsense#L139
https://github.com/Asmageddon/airbreak-plus/blob/master/patch-airsense#L40

(with some minor address fix ups) But had no luck. Do you have 40x ASV firmware to test with?

smhc, 2024-04-10:
I managed to patch the 5cm difference limit for asv-37043-SX567-0305.bin firmware, even though it didn't seem to allow having a difference less than 5 on first install. After messing with some other firmware it eventually worked ok - I think it required the memory erase that it sometimes does when switching firmware.

base64 xdelta patch follows:

```
JVhEWjAwNCUAAAAIABgAFwAAAAAAAAAAAAAAAAAAAABhc3YtMzcwNDMtU1g1NjctMDMwNS5iaW5h
c3YtcHMtbm8tbGltaXQtMzB4LmJpbh+LCAAAAAAAAAM74HaHlYHBCEEwMDAwAkkuBk9GNgYWhi8M
wgwngGKYcgxAOQ0UOQYGAHtlTJhOAAAAH4sIAAAAAAAAA2NgaGBmYGDSaLv99Pe/Caf/bWv7Zvsp
WY+7ocGBkYlHoyCxJDlDISWxJFHz52EVr9OXlhzZL+B9carm60A/RkaJxOIyXWNzAxNj3eAIUzNz
XQNjA1O9pMw8Pwmen5d/cs2JsZgVlipxzBRoGgODMCPDB0agXYyfGLfNBTJYGdfOZxQH0SfnMz6P
ZWCwZXz7l7Ed6BxGxq8NTN2MIEZjE1P3NWkQq/eGrDaI3nlDdipY6vxN2Y3LlACW9/WKwQAAAAAA
AIolWERaMDA0JQ==
```

JohnnyDoe777, 2025-02-21:
@smhc Was it difficult to find out which hex addresses needed to be changed to another values? I wonder if this can be changed for 37113-SX567-0401 ASV fw? How did you erase memory after flash? via the cable connected from the PC and to the device board or from the device menu? Are you using this PS limit less than 5 and is it reliable? That would work for me because now it gives me too much pressure when using nasal pillows.

smhc, 2025-02-22:
@JohnnyDoe777 I believe those edits will work on other ASV firmware revisions. Or at least you could search for the pattern and do the same replacement, or just use the same firmware. Finding the bytes required reverse engineering with the aid of some of the symbols and addresses already discovered under this project.

There's no harm in flashing bad firmware if you've got a good bin to flash if it fails. The "erase" happens on the device under certain conditions, presumably when it detects a firmware version change or some other flag signifying an erase is required. Thus a particular firmware may fail initially, but if you trigger an erase using different firmware, the original one may start working after the erase.

It worked reliably when I used it, but the lower PS didn't have effective therapy vs CPAP, and it was unnecessary once I got used to it.

JohnnyDoe777, 2025-02-22:
@smhc Thanks for the reply and the information provided. I appreciate it very much :thumbsup:

PaulEvo007, 2025-02-28:
@JohnnyDoe777 Yes those edits are working well on ASV firmware versions. Off course memory addresses will differ from one version to an other. For the AirCurve ASV SX584-0305 I had a look at the firmware and changed these in the script, activated extra debugging menu's as well and all is working well. Oh and changed also the color scheme, kind of personalization of the device. I do not have the ASV SX584-0401 firmware b.t.w. but only memory addresses for the patching will differ. 
B.t.w. changing from an APAP to ASV therapy decreased my AHI from 3.5 to 0.1

JohnnyDoe777, 2025-02-28:
Nice. It's a good your AHI so dropped. 
Are you guys using the backup frequency off as well? Do you manage to sleep through the night without waking up? That's my goal that I'm trying to achieve. I also have an AHI of around 0.00 to 0.01, but I still wake up 3 to 5 times a night here and there. I am now using the unmodified 37113-SX567-0401. I don't know if just reducing the difference between PS to less than 5 would be enough, or even turning off the backup frequency. It would be worth a try to see what that does. But unfortunately I don't know how I would derive those addresses that need to be edited in the 0401 firmware.

PaulEvo007, 2025-02-28:
@JohnnyDoe777 I'm very happy with the results and without therapy the AHI was between 15 and 20. The therapist was convinced that the apneas were purely obstructive. I was quite sure that a lot were central, so a complex apnea. Your AHI is more or less perfect and so will your SpO2. Did you monitor that during the night? Within the "apnea industry" about 95 % of the people/patients are in the obstructive category and that's a 'simple' fix, keeping the airway open. Within the category of 5% centraI sleep apnea it gets a lot complexer. Perhaps you are in that small group. For me a high sudden PS is a problem and therefore getting it lower than the standard gives me more relief and less wakening up, it takes some time for the tuning and data analytics in Oscar. If you like I can make the modifications for you so you can do some testing and tuning with a lower PS.

JohnnyDoe777, 2025-02-28:
Did I understand you correctly that you would try to make a modification to the ASV SX584-0401 fw?. Would you really do it, because it's probably not an easy thing to do? I can't ask you to do such a thing.
I also think that the sudden change in PS may be the cause of the wake up. I use pretty low pressures EPAP only 5.2 and PS min 3.2 but PS max is too high for my nasal pillow mask, which has 0 leaks and therefore doesn't need such high PS (13.4 cmH2O). If I could lower the PS max to around 11 cmH2O, it might lead to less awakenings. Some reddit people say that even that backup frequency can cause awakenings. I'm more bothered by the high PS I guess.

PaulEvo007, 2025-02-28:
@JohnnyDoe777 Yes you did understand it correctly and it's fine with me. I know the struggle for finding the right medical direction/therapy and doing some testing with a lower PS by tuning the device parameters will give direction and insights. Especially if you can have a systematic approach with changing the device parameters, data understanding and data analytics.

JohnnyDoe777, 2025-03-01:
@PaulEvo007 can you contact me at impulse9[at]tutamail.com for more instruction? Thank you

MinuteVent, 2025-03-16:
@JohnnyDoe777 I sent you an email. Thanks